PiSugar/PiSugar #2: Where to buy ?

AntoineTurmel, 2019-04-10:
Will this board/battery be sell someday ?

JdaieLin, 2019-04-10:
Hi Antoine, yes it's current available in China. If everything goes fine, you will be able to buy it on Amazon US soon in this month.

AntoineTurmel, 2019-04-10:
Nice ! I look forward it then :)
How does it compete with JuiceBox Zero board ?

JdaieLin, 2019-04-10:
I think the main purposes of designing PiSugar are portability and productization. We came out this idea when we were creating a portable device. In fact, there are solutions like JuiceBox supplying power through gpio pins, but they take too much space. We really need to put the zero and battery into a small box (P.S. We don't know how to model at that time...). So after we created the battery, we build the cases for the battery and also, for other piHats. I think it's important if you send a pi project, as a gift, to a friend who knows little about developing. I hope it looks like a complete product : )

AntoineTurmel, 2019-04-10:
Ok thanks for the explanation 👍

palradu, 2019-04-13:
Q: Is there any way that we could import it from China? Is it available on Aliexpress, Gearbest etc.? I'm asking because I doubt that we'll be able to get it in Europe through Amazon :(

JdaieLin, 2019-04-14:
> Q: Is there any way that we could import it from China? Is it available on Aliexpress, Gearbest etc.? I'm asking because I doubt that we'll be able to get it in Europe through Amazon :(

I see. If we can solve the shipping problem, I think it can will be available on Tindie or Aliexpress too.
In fact, I had tried to mail one of this from the Netherlands to a friend in Germany using postNL, but it didn't arrive... I need some time to find a reliable way to ship this oversea.

palradu, 2019-04-14:
PostNL should work alright. I ordered online quite a few times and more often than not, the items were shipped through PostNL. The only problem in Europe, with a board like this, is the battery. The EU has strict rules when it comes to the import of batteries from overseas and that could make it so the package gets confiscated at customs. A good read on this is here: https://andrewminalto.com/shipping-batteries/ . If there were a way to add a battery plug to the board so we could use locally available batteries, that would simplify the whole process. I'm thingking of a connector like this: https://www.sparkfun.com/products/9749 . All we'd need then would be the battery specs, so we could buy the right type of battery.

JdaieLin, 2019-04-14:
You are right. Sadly I sent a pi zero and a keyboard with the battery... We had considered about battery plug but it takes much space. If the shipping problem exist, maybe we should reconsider about a plug version.

palradu, 2019-04-14:
Indeed, a plug would take up some space, but the advantage is that you could ship the board anywhere in the world then. You could find the smallest plug that works with that type of battery and if it's not that widespread, when you send out the board, you could send the female part of the plug as an extra and we'd just need to attach it to the battery leads with screws. I've been looking for a board like this for a while, and I couldn't find one. And if I were to make one myself, I couldn't fit it in a case. And since I don't know how to 3D model, I couldn't create the case that I needed. Your implementation is perfect and I can't wait to get my hands on the board. Even without the plug. Even if I have to solder the battery leads myself :)

leperjulien, 2019-04-16:
Wow just found about your project on hackaday which led me here :D And you can me add me up to the list of people located in Europe (FR) and who are really interested in trying your board with a RPI Zero W ! Desing is awesome btw

JdaieLin, 2019-04-23:
> Indeed, a plug would take up some space, but the advantage is that you could ship the board anywhere in the world then. You could find the smallest plug that works with that type of battery and if [...]

Where do you usually buy batteries from? Do they have plug already soldered? If so, maybe we should choose a plug type that is widely used.

palradu, 2019-04-23:
Yes, most of them already have the plug soldered on. In Romania, we can get them from local distributors, such as: https://www.robofun.ro/surse_de_alimentare/acumulatori , https://www.optimusdigital.ro/ro/296-acumulatori-lipo or https://www.conexelectronic.ro/ro/417-acumulatori and we can also order from within the EU, Pimoroni being a good example: https://shop.pimoroni.com/collections/prototyping/power . All links point to the available battery packs, so you can maybe get an idea about what would be the best plug to choose.

JdaieLin, 2019-04-23:
Apparently I had never heard of these websites until today... Many thanks!

JdaieLin, 2019-04-23:
> Wow just found about your project on hackaday which led me here :D And you can me add me up to the list of people located in Europe (FR) and who are really interested in trying your board with a RPI Zero W ! Desing is awesome btw

Yeah I notice that you are all from Europe. I will let you know once we solve the shipping problem. Thanks!

palradu, 2019-04-23:
> Apparently I had never heard of these websites until today... Many thanks!

You're welcome! I hope the links will help :)

JdaieLin, 2019-05-04:
Hi everyone. I'm glad to inform you that PiSugar is now available on amazon!

https://www.amazon.com/dp/B07R7Z2334

For the shipping problems, we have found a reliable shipping agent and tried to send the products(including boards and batteries) to Europe countries(NL, FR, GE). It takes about 10 days and all of the packages are successfully received.

JdaieLin, 2019-05-05:
I just know that there will be extra shipping fee when buying things from Amazon US in Europe. Therefore we will soon put them on Amazon Europe, too. Maybe the day after tomorrow. I'll let you know asap.

palradu, 2019-05-05:
This is amazing news! Thank you so much! I can't wait for it to be available on Amazon Europe, so I can get my hands on it :D

JdaieLin, 2019-05-08:
Updated!

https://www.amazon.es/dp/B07RLZ8Q3M
https://www.amazon.it/dp/B07RFVDG3P
https://www.amazon.fr/dp/B07RHX3P9Z
https://www.amazon.de/dp/B07RHWYMTJ
https://www.amazon.co.uk/dp/B07RDNT8CY

JdaieLin, 2019-05-14:
> This is amazing news! Thank you so much! I can't wait for it to be available on Amazon Europe, so I can get my hands on it :D

It turns out that Amazon doesn't have .ro site. So consider buying it from amazon.de. We had told our seller friends to lower the shipping fee in .de for eastern Europe countries.

palradu, 2019-05-21:
> It turns out that Amazon doesn't have .ro site. So consider buying it from amazon.de. We had told our seller friends to lower the shipping fee in .de for eastern Europe countries.

Yup, we still don't have an Amazon.ro site :( But yes, the .de or .es is where I will buy from :) Thank you so much for this! :)

AntoineTurmel, 2019-05-27:
I received mine today 👍

autrion, 2020-01-17:
Has anyone had luck ordering from Amazon? I have ordered from Germany and the first package was lost according to the tracking info. Then the vendor send it again but I have never recieved anything, not even a second tracking number! Slowly I have given up and are looking for an alternative.

StevenMercurio, 2020-02-28:
Anyone have links to where I can buy a case?  Don't have a 3D printer or know anyone who does (unless there is some cloud based 3D printing service I don't know about yet)

JdaieLin, 2020-03-02:
Hi Steven,

you can buy case along with PiSugar battery on Tindie.com. Also recommended a cloud based 3D printing service: http://www.wenext.hk

JdaieLin, 2020-03-09:
> Has anyone had luck ordering from Amazon? I have ordered from Germany and the first package was lost according to the tracking info. Then the vendor send it again but I have never recieved anything, not even a second tracking number! Slowly I have given up and are looking for an alternative.

Hi autrion, 

please contact pisugar.zero@gmail.com for the order issue, thanks!

GSVNoFixedabode, 2021-01-03:
Hi, is there any update on making this available via AliExpress? I tried the Amazon listing but it doesn’t allow shipping to New Zealand. I’d add my vote to a batteryless version!

JdaieLin, 2021-03-06:
> Hi, is there any update on making this available via AliExpress? I tried the Amazon listing but it doesn’t allow shipping to New Zealand. I’d add my vote to a batteryless version!

You can try Tindie.com

closed 2024-09-11 by github-actions[bot]