PiSugar/PiSugar #6: PiSugar 1 and power hungry devices

leperjulien, 2019-10-30:
Hi,
I've been looking for a solution to my problem for a while now and i'd like to ask you if maybe PiSugar could be the answer ?

I'm using a Raspberry Pi Zero W with a HAT attached to it, which require a 2.5A power supply to properly operate (otherwise if using a 1.5A power supply or lower the Raspberry will boot loop) 
OR
i need to use a powered USB hug with a 2.5a power supply as well..

If i use PiSugar 1200mAh with my setup and try to power it from a regular USB port (bus powered) do you think it will work or my issue cannot be solved this way ?

Thanks in advance

JdaieLin, 2019-10-30:
Pisugar1 1200mah can support 2.5A output(max 2.6A tested). But keep outputting such big current is not recommended.

------------------ Original ------------------
From: leperjulien <notifications@github.com&gt;
Date: Wed,Oct 30,2019 0:29 PM
To: PiSugar/PiSugar <PiSugar@noreply.github.com&gt;
Cc: Subscribed <subscribed@noreply.github.com&gt;
Subject: Re: [PiSugar/PiSugar] PiSugar 1 and power hungry devices (#6)

leperjulien, 2019-11-03:
Thanks for your fast answer :) I think i should have been more specific !

My raspberry pi zero W is attached to a GPRS hat (which require a 2.5A power supply to work) but this setup doesn't actually pull 2.5A when connected, i'm trying to power the whole thing from a single USB port but i've added a small USB voltage tester and it indicates that it's using 0.4A when plugged in, the GPRS module only need a 2.5A for really short time when it power up or make a voice call !

Do you think adding PiSugar to my setup willl allow to power it from a single USB port ?

Also i'm not sure this can be done but i'll ask anyway !

Can i use PiSugar this way ? Connected to a hub while powering the raspberry ?

[https://i.ibb.co/z7t37NL/pisugarpng.jpg](url)

Sorry if my question aren't very clear, english isn't my native tongue

Thanks in advance

JdaieLin, 2019-11-03:
Emmm... If the GPRS hat and pizero together only need a 2.5A for a short time then PiSugar is ok.

You can also connect it to a host computer in the way that you described. 

However when there is external power (from the host usb hub), the PiSugar chip will go sleep since the host power is enough for the pi and phat. If you want to keep the pi on, just click to turn on PiSugar before you unplug from the computer. 

So I think you need to make sure whether your pi and GPRS can work while just connecting to the usb hub. If it works fine, then adding a PiSugar is no problem.

closed 2021-03-06 by JdaieLin