PiSugar/PiSugar #7: Parameters for 3D Printing the case

AntoineTurmel, 2019-11-07:
Hello, what are the best parameters to get a clean 3D print of the case?
- Rafts ? Using PVA ?
- Supports ?
- Resolution ?
- Infill % ?
- Material ?
- 3D Printer you use ?

AntoineTurmel, 2019-11-07:
I'll try with:
- PVA Rafts
- Supports (plate adhesion)
- 0.1mm resolution
- 20% Infill
- Tough PLA Black
- Printing with Ultimaker 3 Extended

JdaieLin, 2019-11-07:
Hi, Antonie. We usually commission a professional factory to print these cases. 
Here's some detail I can provide:

- They use large industrial SLA printers (UnionTech or HP)
- STL model tolerances and angles usually set to 0.0045mm and 6.333
- Material: Somos GP resin (tolerances 200μm)

You can get a quote by uploading the STL files to their website: https://www.wenext.hk. I am not sure about their oversea shipping fee. 

If you would like to get a ready to use case from us, please contact pisugar.zero@gmail.com.

JdaieLin, 2019-11-07:
> I'll try with:
> 
> * PVA Rafts
> * Supports (plate adhesion)
> * 0.1mm resolution
> * 20% Infill
> * Tough PLA Black
> * Printing with Ultimaker 3 Extended

> I'll try with:
> 
> * PVA Rafts
> * Supports (plate adhesion)
> * 0.1mm resolution
> * 20% Infill
> * Tough PLA Black
> * Printing with Ultimaker 3 Extended

emmm... Ultimaker 3 Extended seems really good in resolution, if it can keep 0.1mm resolution I think it would be ok.

AntoineTurmel, 2019-11-07:
Thanks, it's currently printing :)
I want to be sure that Pi0/PiSugar fits well, because I'll make a custom case based on it.

AntoineTurmel, 2019-11-08:
Is it normal that the pisugar_case_button is at the same level of the case shell ?
It should stand out a little bit outside for the user to be able to push it.
I'm trying a modified version of it with a bigger size.

AntoineTurmel, 2019-11-08:
Here is a photo of the case printed with a bigger button:
[image]
[image]

Here is the STL file modified:
[pisugar_case_button_bigger.zip](https://github.com/PiSugar/PiSugar/files/3825357/pisugar_case_button_bigger.zip)

I have yet to try it with the board inside, but I'm confident 😉

AntoineTurmel, 2019-11-09:
So I finally assembled my case at home, it's working well but there is a bit of looseness.
(pisugar_case_thin_shell.stl file)

Here are some pictures:
[image]
[image]
[image]

It looks a bit dirty, I have to redo the cover but I'm quite happy for now 😄

JdaieLin, 2019-11-09:
Nice work! =_<

snarflakes, 2021-06-01:
For these 3d printed cases, do the metal pi ports come through the printed material? Or lie just inside of the printed material? Thanks

closed 2024-09-11 by github-actions[bot]