PiSugar/PiSugar #10: Unespected behaviour (or dumb user, I don't know...)

mickymix, 2020-02-08:
I've attached my 1200mah PiSugar to my Pi0W which I use in the car.

I would like to use it as a backup power for the Pi0w when the car removes power from the 12V circuit some minutes after having closed it.

Since I only have one USB slot available, I'm using a Y cable to split power and data (https://www.amazon.it/gp/product/B071H54CKS=).
I've attached the data side to the USB on the Pi0w and the power side to the PiSugar.

The strange thing that happens is that, whenever the car removes the power from the USB port, the Pi0w powers off, as the PiSugar, while being charged, shuts down too...

I've tried connecting the power side of the Y cable to the Pi0w, but this doesn't charge the PiSugar, so it only lasts until there is power in the battery...

Is this an intended behaviour ?
Where is my error ?

JdaieLin, 2020-02-08:
Hi Micky,
I cannot open the link: https://www.amazon.it/gp/product/B071H54CKS=
had you met this problem while using other USB cable?

JdaieLin, 2020-02-08:
Connecting the power side of the Y cable to the Pi0w will not charge the PiSugar. It's normal behavior.

I'm wondering if the cable can charge the PiSugar along without the data side connecting to the pi0w. Not sure how it actually split power and data.

mickymix, 2020-02-08:
Hi Jdaie,
sorry, the link for cable was wrong (added one character by mistake...).

Anyway, the cable was suggested by other users using the Pi0w for my same needs on this other GH project: https://github.com/marcone/teslausb
The cable has been perfectly working before I attached the PiSugar to my Pi0w, so I believe it might not be the problem... (will check anyway getting 12V with another cable elsewhere...).

The cable is capable of charging the PiSugar, as when car's 12V is present, Pi and PiSugar are working, and PiSugar is always well charged (4 leds full on...).
Anyway when the car cuts 12V, the PiSugar switches off, while still having fully loaded battery.

Indeed, if I manually switch the PiSugar on, it powers the Pi correctly.

It seems that when power disappears from the cable, this makes the PiSugar switch off too, and this seems strange to me, unless this might be "cable-related".

Could the PiSugar behave differently when incoming power is removed gradually or suddenly ?
Should the PiSugar be already on before adding incoming power to have it stay on when it is removed ?
I'll try some more cable and switch-on combinations as soon as I can, but just wanted to investigate if there is anything I should know before.

Thank you for your help.

JdaieLin, 2020-02-08:
It's strange that the PiSugar switches off. Normally it will keep on when removing the power, switching power source from USB to battery. And the pi0w will reboot during the switching, rather than turning off.

JdaieLin, 2020-02-08:
There is no need to turn on PiSugar before adding incoming power. It doesn't matter.

mickymix, 2020-02-08:
Thank you for your fast reply.

I also had the chance to make a couple of checks with the cables.

I discovered that this strange behaviour ALWAYS happens with the Y splitter cable: either when I remove the cable from the supply side, either when I remove the cable on the power exit side...
So I thought the problem was the cable indeed...

Tried working without the splitter, and I've been successfull for a couple of minutes... :-(

Not more than a couple of minutes, because on the first 3 or 4 trials I had success: the PiSugar did not switch off and continued supplying power to the pi0w, which did not reboot...

But after few minutes I did remove the power coming from the car and it happened again...

FYI, the PiSugar was not completely charged, but was around 50% (two leds lit)

I think I'll need to further investigate, because the above tests were made without waiting for the pi0w to complete bootstraps (just looking at the working green led...).
To be more precise I think I'll need to have some more spare time.

I'll let you know.

JdaieLin, 2020-02-08:
There is no difference when removing the cable on supply side or power exit side.
On and off is decided by the power management chip on board.

1. I am not sure the cable has strictly split the power and data, perhaps the data side also has power output? If so, then the PiSugar will work inside a close circus which will leads to unknown result.
2. maybe the chip is somehow damaged thus it behaves strangely.

I suggest you test it on other simple usb cables to see whether it behaves the same.

HynekPetrak, 2020-04-07:
Hi, I have a rpi0w with a waveshare eink 2.13" V2 display attached. Just received my sugar from tindie and mounted. I charged the pisugar to 4 blue leds. I observe two things:
- as soon as I disconnect the pisugar from the USB power chager, the rpi0w reboots. Everytime, tested with two cables.
- when I'm connected to a USB port on rpi0w, the pisugar turns off. When I disconnect from Rpi USB port, the devices shuts down and I need to turn on pisegar with the button.

JdaieLin, 2020-04-07:
> Hi, I have a rpi0w with a waveshare eink 2.13" V2 display attached. Just received my sugar from tindie and mounted. I charged the pisugar to 4 blue leds. I observe two things:
>
> * as soon as I [...]

The reboot issue has nothing to do with the cable, it's due to the powerIC design. It will reboot the pi when switching from USB power to battery.

For the second point I guess you were connecting pi0w to an USB host (like a PC). The host will power the device which are connecting it. When there's external power source, pisugar might not able to detect power load and thus shutdown automatically. If you need to keep pi0w on when  disconnecting from the host, just turn the pisugar on before you unplug the USB.

closed 2021-03-06 by JdaieLin