PiSugar/PiSugar #12: Trouble installing power manager on PiSugar2

jzting, 2020-07-30:
After running the command to install, the installation seems to complete successfully but nothing is accessible at http://192.168.1.8:8421

```
pi@OpenframeRaspi:~ $ curl http://cdn.pisugar.com/release/Pisugar-power-manager.sh | sudo bash
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  2116  100  2116    0     0    813      0  0:00:02  0:00:02 --:--:--  1195
Download PiSugar-server and PiSugar-poweroff package
--2020-07-29 20:25:39--  http://cdn.pisugar.com/release/pisugar-server_1.4.3_armhf.deb
Resolving cdn.pisugar.com (cdn.pisugar.com)... 101.227.0.140, 101.227.0.139, 101.227.0.138, ...
Connecting to cdn.pisugar.com (cdn.pisugar.com)|101.227.0.140|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1298636 (1.2M) [application/x-debian-package]
Saving to: ‘/tmp/pisugar-server_1.4.3_armhf.deb’

/tmp/pisugar-server_1.4.3_armhf.deb                                                        100%[======================================================================================================================================================================================================================================>]   1.24M   103KB/s    in 13s

2020-07-29 20:25:54 (97.1 KB/s) - ‘/tmp/pisugar-server_1.4.3_armhf.deb’ saved [1298636/1298636]

--2020-07-29 20:25:54--  http://cdn.pisugar.com/release/pisugar-poweroff_1.4.3_armhf.deb
Resolving cdn.pisugar.com (cdn.pisugar.com)... 101.227.0.133, 101.227.0.140, 101.227.0.139, ...
Connecting to cdn.pisugar.com (cdn.pisugar.com)|101.227.0.133|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 249438 (244K) [application/x-debian-package]
Saving to: ‘/tmp/pisugar-poweroff_1.4.3_armhf.deb’

/tmp/pisugar-poweroff_1.4.3_armhf.deb                                                      100%[======================================================================================================================================================================================================================================>] 243.59K  81.8KB/s    in 3.0s

2020-07-29 20:25:58 (81.8 KB/s) - ‘/tmp/pisugar-poweroff_1.4.3_armhf.deb’ saved [249438/249438]

Open I2C Interface
Uninstall old packages if installed
(Reading database ... 92130 files and directories currently installed.)
Removing pisugar-server (1.4.3) ...
Removing pisugar-poweroff (1.4.3) ...
Install packages
Selecting previously unselected package pisugar-server.
(Reading database ... 92113 files and directories currently installed.)
Preparing to unpack .../pisugar-server_1.4.3_armhf.deb ...
Unpacking pisugar-server (1.4.3) ...
Selecting previously unselected package pisugar-poweroff.
Preparing to unpack .../pisugar-poweroff_1.4.3_armhf.deb ...
Unpacking pisugar-poweroff (1.4.3) ...
Setting up pisugar-server (1.4.3) ...
Setting up pisugar-poweroff (1.4.3) ...
Clean up
Now navigate to http://192.168.1.8:8421 on your browser to see PiSugar power management
If you have any question,please feel free to contact us.
The PiSugar Team https://www.pisugar.com
```

JdaieLin, 2020-07-30:
Please try these commands to start/stop pisugar-server:

# reload daemon
sudo systemctl daemon-reload

# check status
sudo systemctl status pisugar-server

# start service
sudo systemctl start pisugar-server

# stop service
sudo systemctl stop pisugar-server

# disable service
sudo systemctl disable pisugar-server

# enable service
sudo systemctl enable pisugar-server

jzting, 2020-07-30:
It looks like it's having trouble initializing:

```
pi@OpenframeRaspi:~ $ sudo systemctl status pisugar-server
● pisugar-server.service - pisugar-server
   Loaded: loaded (/lib/systemd/system/pisugar-server.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2020-07-30 09:15:46 EDT; 1min 2s ago
 Main PID: 1080 (pisugar-server)
   Memory: 1008.0K
   CGroup: /system.slice/pisugar-server.service
           └─1080 /usr/bin/pisugar-server --config config.json --model PiSugar 2 (4-LEDs) --web /usr/share/pisugar-server/web --http 0.0.0.0:8421 --ws 0.0.0.0:8422 --tcp 0.0.0.0:8423 --uds /tmp/pisugar-server.sock

Jul 30 09:16:35 OpenframeRaspi pisugar-server[1080]: [2020-07-30T13:16:35Z WARN  pisugar_core] Load configuration failed, auto recovery...
Jul 30 09:16:35 OpenframeRaspi pisugar-server[1080]: [2020-07-30T13:16:35Z ERROR pisugar_server] PiSugar init failed
Jul 30 09:16:38 OpenframeRaspi pisugar-server[1080]: [2020-07-30T13:16:38Z WARN  pisugar_core] Load configuration failed, auto recovery...
Jul 30 09:16:38 OpenframeRaspi pisugar-server[1080]: [2020-07-30T13:16:38Z ERROR pisugar_server] PiSugar init failed
Jul 30 09:16:41 OpenframeRaspi pisugar-server[1080]: [2020-07-30T13:16:41Z WARN  pisugar_core] Load configuration failed, auto recovery...
Jul 30 09:16:41 OpenframeRaspi pisugar-server[1080]: [2020-07-30T13:16:41Z ERROR pisugar_server] PiSugar init failed
Jul 30 09:16:44 OpenframeRaspi pisugar-server[1080]: [2020-07-30T13:16:44Z WARN  pisugar_core] Load configuration failed, auto recovery...
Jul 30 09:16:44 OpenframeRaspi pisugar-server[1080]: [2020-07-30T13:16:44Z ERROR pisugar_server] PiSugar init failed
Jul 30 09:16:47 OpenframeRaspi pisugar-server[1080]: [2020-07-30T13:16:47Z WARN  pisugar_core] Load configuration failed, auto recovery...
Jul 30 09:16:47 OpenframeRaspi pisugar-server[1080]: [2020-07-30T13:16:47Z ERROR pisugar_server] PiSugar init failed```

JdaieLin, 2020-07-30:
I need to see what happened in the detail log.
Try this command:
journalctl -u pisugar-server

 

 pi@OpenframeRaspi:~ $ curl http://cdn.pisugar.com/release/Pisugar-power-manager.sh | sudo bash   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current                                  Dload  Upload   Total   Spent    Left  Speed 100  2116  100  2116    0     0    813      0  0:00:02  0:00:02 --:--:--  1195 Download PiSugar-server and PiSugar-poweroff package --2020-07-29 20:25:39--  http://cdn.pisugar.com/release/pisugar-server_1.4.3_armhf.deb Resolving cdn.pisugar.com (cdn.pisugar.com)... 101.227.0.140, 101.227.0.139, 101.227.0.138, ... Connecting to cdn.pisugar.com (cdn.pisugar.com)|101.227.0.140|:80... connected. HTTP request sent, awaiting response... 200 OK Length: 1298636 (1.2M) [application/x-debian-package] Saving to: ‘/tmp/pisugar-server_1.4.3_armhf.deb’ /tmp/pisugar-server_1.4.3_armhf.deb                                                        100%[======================================================================================================================================================================================================================================&gt;]   1.24M   103KB/s    in 13s 2020-07-29 20:25:54 (97.1 KB/s) - ‘/tmp/pisugar-server_1.4.3_armhf.deb’ saved [1298636/1298636] --2020-07-29 20:25:54--  http://cdn.pisugar.com/release/pisugar-poweroff_1.4.3_armhf.deb Resolving cdn.pisugar.com (cdn.pisugar.com)... 101.227.0.133, 101.227.0.140, 101.227.0.139, ... Connecting to cdn.pisugar.com (cdn.pisugar.com)|101.227.0.133|:80... connected. HTTP request sent, awaiting response... 200 OK Length: 249438 (244K) [application/x-debian-package] Saving to: ‘/tmp/pisugar-poweroff_1.4.3_armhf.deb’ /tmp/pisugar-poweroff_1.4.3_armhf.deb                                                      100%[======================================================================================================================================================================================================================================&gt;] 243.59K  81.8KB/s    in 3.0s 2020-07-29 20:25:58 (81.8 KB/s) - ‘/tmp/pisugar-poweroff_1.4.3_armhf.deb’ saved [249438/249438] Open I2C Interface Uninstall old packages if installed (Reading database ... 92130 files and directories currently installed.) Removing pisugar-server (1.4.3) ... Removing pisugar-poweroff (1.4.3) ... Install packages Selecting previously unselected package pisugar-server. (Reading database ... 92113 files and directories currently installed.) Preparing to unpack .../pisugar-server_1.4.3_armhf.deb ... Unpacking pisugar-server (1.4.3) ... Selecting previously unselected package pisugar-poweroff. Preparing to unpack .../pisugar-poweroff_1.4.3_armhf.deb ... Unpacking pisugar-poweroff (1.4.3) ... Setting up pisugar-server (1.4.3) ... Setting up pisugar-poweroff (1.4.3) ... Clean up Now navigate to http://192.168.1.8:8421 on your browser to see PiSugar power management If you have any question,please feel free to contact us. The PiSugar Team https://www.pisugar.com

fengyc, 2020-07-30:
@jzting Try the latest Pre-Release build https://github.com/PiSugar/pisugar-power-manager-rs/releases . The configuration file  /etc/pisugar-server/config.json is corrupted , and you need to delete it. 

```
sudo rm -f /etc/pisugar-server/config.json
sudo systemctl restart pisugar-server
```

jzting, 2020-07-30:
Is there a one-liner I can run to get the pre-release build?

fengyc, 2020-07-31:
```
wget https://github.com/PiSugar/pisugar-power-manager-rs/releases/download/latest/pisugar-server_1.4.4_armhf.deb
sudo dpkg -i pisugar-server_1.4.4_armhf.deb
```
OR
```
wget http://cdn.pisugar.com/release/Pisugar-power-manager.sh
bash Pisugar-power-manager.sh -c nightly -v 1.4.4
```

jzting, 2020-07-31:
I've done this but still get an error:
```
pi@OpenframeRaspi:~ $ wget https://github.com/PiSugar/pisugar-power-manager-rs/releases/download/latest/pisugar-server_1.4.4_armhf.deb
--2020-07-31 11:48:24--  https://github.com/PiSugar/pisugar-power-manager-rs/releases/download/latest/pisugar-server_1.4.4_armhf.deb
Resolving github.com (github.com)... 140.82.113.3
Connecting to github.com (github.com)|140.82.113.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/231610023/f5f43000-d2c6-11ea-8a89-58ed90df7737?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20200731%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20200731T154831Z&X-Amz-Expires=300&X-Amz-Signature=5c90967d0b9d4e98f03895bfeeb46e4576b803eb24ff42ac3e6cd049dde24a42&X-Amz-SignedHeaders=host&actor_id=0&repo_id=231610023&response-content-disposition=attachment%3B%20filename%3Dpisugar-server_1.4.4_armhf.deb&response-content-type=application%2Foctet-stream [following]
--2020-07-31 11:48:31--  https://github-production-release-asset-2e65be.s3.amazonaws.com/231610023/f5f43000-d2c6-11ea-8a89-58ed90df7737?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20200731%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20200731T154831Z&X-Amz-Expires=300&X-Amz-Signature=5c90967d0b9d4e98f03895bfeeb46e4576b803eb24ff42ac3e6cd049dde24a42&X-Amz-SignedHeaders=host&actor_id=0&repo_id=231610023&response-content-disposition=attachment%3B%20filename%3Dpisugar-server_1.4.4_armhf.deb&response-content-type=application%2Foctet-stream
Resolving github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)... 52.217.41.132
Connecting to github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)|52.217.41.132|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1296312 (1.2M) [application/octet-stream]
Saving to: ‘pisugar-server_1.4.4_armhf.deb’

pisugar-server_1.4.4_armhf.deb                                                             100%[========================================================================================================================================================================================================================================>]   1.24M   147KB/s    in 8.5s

2020-07-31 11:48:40 (150 KB/s) - ‘pisugar-server_1.4.4_armhf.deb’ saved [1296312/1296312]

pi@OpenframeRaspi:~ $ sudo dpkg -i pisugar-server_1.4.4_armhf.deb
(Reading database ... 92324 files and directories currently installed.)
Preparing to unpack pisugar-server_1.4.4_armhf.deb ...
Unpacking pisugar-server (1.4.4) over (1.4.3) ...
Setting up pisugar-server (1.4.4) ...

Configuration file '/etc/pisugar-server/config.json'
 ==> Deleted (by you or by a script) since installation.
 ==> Package distributor has shipped an updated version.
   What would you like to do about it ?  Your options are:
    Y or I  : install the package maintainer's version
    N or O  : keep your currently-installed version
      D     : show the differences between the versions
      Z     : start a shell to examine the situation
 The default action is to keep your current version.
*** config.json (Y/I/N/O/D/Z) [default=N] ? y
Installing new version of config file /etc/pisugar-server/config.json ...

pi@OpenframeRaspi:~ $ sudo systemctl status pisugar-server
● pisugar-server.service - pisugar-server
   Loaded: loaded (/lib/systemd/system/pisugar-server.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2020-07-31 11:51:13 EDT; 56s ago
 Main PID: 3791 (pisugar-server)
   Memory: 2.3M
   CGroup: /system.slice/pisugar-server.service
           └─3791 /usr/bin/pisugar-server --config config.json --model PiSugar 2 (4-LEDs) --web /usr/share/pisugar-server/web --http 0.0.0.0:8421 --ws 0.0.0.0:8422 --tcp 0.0.0.0:8423 --uds /tmp/pisugar-server.sock

Jul 31 11:51:56 OpenframeRaspi pisugar-server[3791]: [2020-07-31T15:51:56Z WARN  pisugar_core] Load configuration failed, auto recovery...
Jul 31 11:51:56 OpenframeRaspi pisugar-server[3791]: [2020-07-31T15:51:56Z ERROR pisugar_server] PiSugar init failed: I/O error: No such file or directory (os error 2)
Jul 31 11:51:59 OpenframeRaspi pisugar-server[3791]: [2020-07-31T15:51:59Z WARN  pisugar_core] Load configuration failed, auto recovery...
Jul 31 11:51:59 OpenframeRaspi pisugar-server[3791]: [2020-07-31T15:51:59Z ERROR pisugar_server] PiSugar init failed: I/O error: No such file or directory (os error 2)
Jul 31 11:52:02 OpenframeRaspi pisugar-server[3791]: [2020-07-31T15:52:02Z WARN  pisugar_core] Load configuration failed, auto recovery...
Jul 31 11:52:02 OpenframeRaspi pisugar-server[3791]: [2020-07-31T15:52:02Z ERROR pisugar_server] PiSugar init failed: I/O error: No such file or directory (os error 2)
Jul 31 11:52:05 OpenframeRaspi pisugar-server[3791]: [2020-07-31T15:52:05Z WARN  pisugar_core] Load configuration failed, auto recovery...
Jul 31 11:52:05 OpenframeRaspi pisugar-server[3791]: [2020-07-31T15:52:05Z ERROR pisugar_server] PiSugar init failed: I/O error: No such file or directory (os error 2)
Jul 31 11:52:08 OpenframeRaspi pisugar-server[3791]: [2020-07-31T15:52:08Z WARN  pisugar_core] Load configuration failed, auto recovery...
Jul 31 11:52:08 OpenframeRaspi pisugar-server[3791]: [2020-07-31T15:52:08Z ERROR pisugar_server] PiSugar init failed: I/O error: No such file or directory (os error 2)
```

fengyc, 2020-08-03:
@jzting Could you show me what's inside `/lib/systemd/system/pisugar-server.service`.

It might be the working directory is incorrenct , or the `/etc/pisugar-server` directory doesn't exist.

fengyc, 2020-08-03:
@jzting Please report the following updates here: https://github.com/PiSugar/pisugar-power-manager-rs/issues/13

boris-create, 2020-08-26:
hello i am having a very similar issue with mine as well. command and install completed with no errors however i cant access via: http://192.168.44.44:8421 (my ip address according to ifconfig)...

root@waifu4u:/etc/pwnagotchi# sudo systemctl status pisugar-server
● pisugar-server.service - pisugar-server
   Loaded: loaded (/lib/systemd/system/pisugar-server.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2020-08-26 06:17:18 BST; 55s ago
 Main PID: 16712 (pisugar-server)
   CGroup: /system.slice/pisugar-server.service
           └─16712 /usr/bin/pisugar-server --config config.json --model PiSugar 2 (4-LEDs) --web /usr/share/pisugar-server/web --http 0.0.0.0:8421 --ws 0.0.0.

Aug 26 06:18:00 waifu4u pisugar-server[16712]: [2020-08-26T05:18:00Z WARN  pisugar_core] Load configuration failed, auto recovery...
Aug 26 06:18:00 waifu4u pisugar-server[16712]: [2020-08-26T05:18:00Z ERROR pisugar_server] PiSugar init failed
Aug 26 06:18:03 waifu4u pisugar-server[16712]: [2020-08-26T05:18:03Z WARN  pisugar_core] Load configuration failed, auto recovery...
Aug 26 06:18:03 waifu4u pisugar-server[16712]: [2020-08-26T05:18:03Z ERROR pisugar_server] PiSugar init failed
Aug 26 06:18:06 waifu4u pisugar-server[16712]: [2020-08-26T05:18:06Z WARN  pisugar_core] Load configuration failed, auto recovery...
Aug 26 06:18:06 waifu4u pisugar-server[16712]: [2020-08-26T05:18:06Z ERROR pisugar_server] PiSugar init failed
Aug 26 06:18:09 waifu4u pisugar-server[16712]: [2020-08-26T05:18:09Z WARN  pisugar_core] Load configuration failed, auto recovery...
Aug 26 06:18:09 waifu4u pisugar-server[16712]: [2020-08-26T05:18:09Z ERROR pisugar_server] PiSugar init failed
Aug 26 06:18:12 waifu4u pisugar-server[16712]: [2020-08-26T05:18:12Z WARN  pisugar_core] Load configuration failed, auto recovery...
Aug 26 06:18:12 waifu4u pisugar-server[16712]: [2020-08-26T05:18:12Z ERROR pisugar_server] PiSugar init failed

mandreko, 2020-08-31:
I was having this problem on my pwnagotchi as well. Upgrading to 1.4.5 worked for me.

```
wget https://github.com/PiSugar/pisugar-power-manager-rs/releases/download/v1.4.5/pisugar-poweroff_1.4.5_armhf.deb
wget https://github.com/PiSugar/pisugar-power-manager-rs/releases/download/v1.4.5/pisugar-server_1.4.5_armhf.deb
sudo dpkg -i pisugar-*.deb
```

dirtdiver99, 2020-09-10:
> I was having this problem on my pwnagotchi as well. Upgrading to 1.4.5 worked for me.
>
> ```
> wget [...]

Thank you! I installed 1.4.6 and now I can finally access the PiSugar power management UI web interface. 

However, it doesn't show any battery info. I did notice i2c 0x32 is active but not 0x75. I am using the PiSugar2 with the Waveshare v2 2.13 display, if that might be causing the issue?

fengyc, 2020-09-11:
It looks like the PiSugar2 has been shutdown. 

There is a known issue. When the rpi gets electricity from micro usb,  the PiSugar2 will be under light load ( less than 100mA ). Then the chips will shutdown, and I2C 0x75 will be gone.

tisboyo, 2020-09-11:
I had the same issue and fixed it by cutting the 5v wire on a micro-usb cable, leaving only the two data and ground. This let's you communicate with the pi without causing the pisugar to shut down.

pchois, 2021-11-07:
Hello,

I'm having trouble loading the power management dashboard. I have a Pi Zero running Pi-Star 4.1.5 and a PiSugar 2 (2 LEDs), ran through the setup process and everything was successful but the dashboard won't load. Following this thread, the server seems to be up and running

```
pi-star@pi-star(rw):~$ journalctl -u pisugar-server
-- Logs begin at Wed 2021-11-03 16:44:24 EDT, end at Sun 2021-11-07 12:40:10 EST. --
Nov 03 16:44:47 pi-star systemd[1]: Started pisugar-server.
Nov 03 16:44:47 pi-star pisugar-server[270]: [2021-11-03T20:44:47Z INFO  pisugar_server] TCP listening...
Nov 03 16:44:47 pi-star pisugar-server[270]: [2021-11-03T20:44:47Z INFO  pisugar_server] WS listening...
Nov 03 16:44:47 pi-star pisugar-server[270]: [2021-11-03T20:44:47Z INFO  pisugar_server] UDS listening...
Nov 03 16:44:47 pi-star pisugar-server[270]: [2021-11-03T20:44:47Z INFO  pisugar_server] Http web server listening...
Nov 03 16:44:48 pi-star pisugar-server[270]: Error: Os { code: 30, kind: Other, message: "Read-only file system" }
Nov 03 16:44:48 pi-star systemd[1]: pisugar-server.service: Main process exited, code=exited, status=1/FAILURE
Nov 03 16:44:48 pi-star systemd[1]: pisugar-server.service: Failed with result 'exit-code'.
Nov 03 16:44:58 pi-star systemd[1]: pisugar-server.service: Service RestartSec=10s expired, scheduling restart.
Nov 03 16:44:58 pi-star systemd[1]: pisugar-server.service: Scheduled restart job, restart counter is at 1.
Nov 03 16:44:58 pi-star systemd[1]: Stopped pisugar-server.
Nov 03 16:44:58 pi-star systemd[1]: Started pisugar-server.
Nov 03 16:44:58 pi-star pisugar-server[583]: [2021-11-03T20:44:58Z INFO  pisugar_server] TCP listening...
Nov 03 16:44:58 pi-star pisugar-server[583]: [2021-11-03T20:44:58Z INFO  pisugar_server] WS listening...
Nov 03 16:44:58 pi-star pisugar-server[583]: [2021-11-03T20:44:58Z INFO  pisugar_server] UDS listening...
Nov 03 16:44:58 pi-star pisugar-server[583]: [2021-11-03T20:44:58Z INFO  pisugar_server] Http web server listening...
Nov 03 16:44:59 pi-star pisugar-server[583]: Error: Os { code: 30, kind: Other, message: "Read-only file system" }
Nov 03 16:44:59 pi-star systemd[1]: pisugar-server.service: Main process exited, code=exited, status=1/FAILURE
Nov 03 16:44:59 pi-star systemd[1]: pisugar-server.service: Failed with result 'exit-code'.
Nov 03 16:45:17 pi-star systemd[1]: pisugar-server.service: Service RestartSec=10s expired, scheduling restart.
Nov 03 16:45:17 pi-star systemd[1]: pisugar-server.service: Scheduled restart job, restart counter is at 2.
Nov 03 16:45:17 pi-star systemd[1]: Stopped pisugar-server.
Nov 03 16:45:17 pi-star systemd[1]: Started pisugar-server.
Nov 03 16:45:17 pi-star pisugar-server[1082]: [2021-11-03T20:45:17Z INFO  pisugar_server] TCP listening...
Nov 03 16:45:17 pi-star pisugar-server[1082]: [2021-11-03T20:45:17Z INFO  pisugar_server] WS listening...
Nov 03 16:45:17 pi-star pisugar-server[1082]: [2021-11-03T20:45:17Z INFO  pisugar_server] UDS listening...
Nov 03 16:45:17 pi-star pisugar-server[1082]: [2021-11-03T20:45:17Z INFO  pisugar_server] Http web server listening...
Nov 03 16:45:17 pi-star pisugar-server[1082]: Error: Os { code: 30, kind: Other, message: "Read-only file system" }
Nov 03 16:45:18 pi-star systemd[1]: pisugar-server.service: Main process exited, code=exited, status=1/FAILURE
Nov 03 16:45:18 pi-star systemd[1]: pisugar-server.service: Failed with result 'exit-code'.
Nov 07 12:22:07 pi-star systemd[1]: pisugar-server.service: Service RestartSec=10s expired, scheduling restart.
Nov 07 12:22:07 pi-star systemd[1]: pisugar-server.service: Scheduled restart job, restart counter is at 3.
Nov 07 12:22:07 pi-star systemd[1]: Stopped pisugar-server.
Nov 07 12:22:07 pi-star systemd[1]: Started pisugar-server.
Nov 07 12:22:07 pi-star pisugar-server[1219]: [2021-11-07T17:22:07Z INFO  pisugar_server] TCP listening...
Nov 07 12:22:07 pi-star pisugar-server[1219]: [2021-11-07T17:22:07Z INFO  pisugar_server] WS listening...
Nov 07 12:22:08 pi-star pisugar-server[1219]: [2021-11-07T17:22:08Z INFO  pisugar_server] UDS listening...
Nov 07 12:22:08 pi-star pisugar-server[1219]: [2021-11-07T17:22:08Z INFO  pisugar_server] Http web server listening...
Nov 07 12:23:25 pi-star systemd[1]: Stopping pisugar-server...
Nov 07 12:23:25 pi-star systemd[1]: pisugar-server.service: Main process exited, code=killed, status=15/TERM
Nov 07 12:23:25 pi-star systemd[1]: pisugar-server.service: Succeeded.
Nov 07 12:23:25 pi-star systemd[1]: Stopped pisugar-server.
Nov 07 12:27:01 pi-star systemd[1]: Started pisugar-server.
Nov 07 12:27:01 pi-star pisugar-server[4646]: [2021-11-07T17:27:01Z INFO  pisugar_server] TCP listening...
Nov 07 12:27:01 pi-star pisugar-server[4646]: [2021-11-07T17:27:01Z INFO  pisugar_server] WS listening...
Nov 07 12:27:01 pi-star pisugar-server[4646]: [2021-11-07T17:27:01Z INFO  pisugar_server] UDS listening...
Nov 07 12:27:01 pi-star pisugar-server[4646]: [2021-11-07T17:27:01Z INFO  pisugar_server] Http web server listening...
pi-star@pi-star(rw):~$ sudo systemctl status pisugar-server
● pisugar-server.service - pisugar-server
   Loaded: loaded (/lib/systemd/system/pisugar-server.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2021-11-07 12:27:01 EST; 18min ago
 Main PID: 4646 (pisugar-server)
    Tasks: 3 (limit: 991)
   CGroup: /system.slice/pisugar-server.service
[... cut ...]
Nov 07 12:27:01 pi-star pisugar-server[4646]: [2021-11-07T17:27:01Z INFO  pisugar_server] TCP listening...
Nov 07 12:27:01 pi-star pisugar-server[4646]: [2021-11-07T17:27:01Z INFO  pisugar_server] WS listening...
Nov 07 12:27:01 pi-star pisugar-server[4646]: [2021-11-07T17:27:01Z INFO  pisugar_server] UDS listening...
Nov 07 12:27:01 pi-star pisugar-server[4646]: [2021-11-07T17:27:01Z INFO  pisugar_server] Http web server listening...
```

Pi-Star makes the system read-only by default, do I need to disable that? Not sure that is required since the status reports it's listening, any ideas?

fengyc, 2021-11-08:
> Hello,
>
> I'm having trouble loading the power management dashboard. I have a Pi Zero running Pi-Star 4.1.5 and a PiSugar 2 (2 LEDs), ran through the setup process and everything was successful but [...]

@pchois The latest development build of pisugar-server should work,   see https://github.com/PiSugar/pisugar-power-manager-rs/releases

pchois, 2021-11-08:
> @pchois The latest development build of pisugar-server should work, see https://github.com/PiSugar/pisugar-power-manager-rs/releases

no change 

```
Download PiSugar-server and PiSugar-poweroff package
--2021-11-08 08:12:28--  http://cdn.pisugar.com/release/pisugar-server_1.6.1_armhf.deb
Resolving cdn.pisugar.com (cdn.pisugar.com)... 47.246.23.197
Connecting to cdn.pisugar.com (cdn.pisugar.com)|47.246.23.197|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1341232 (1.3M) [application/x-debian-package]
Saving to: ‘/tmp/pisugar-server_1.6.1_armhf.deb’

/tmp/pisugar-server_1.6.1_armhf.deb                      100%[================================================================================================================================>]   1.28M   987KB/s    in 1.3s

2021-11-08 08:12:31 (987 KB/s) - ‘/tmp/pisugar-server_1.6.1_armhf.deb’ saved [1341232/1341232]

--2021-11-08 08:12:31--  http://cdn.pisugar.com/release/pisugar-poweroff_1.6.1_armhf.deb
Resolving cdn.pisugar.com (cdn.pisugar.com)... 47.246.23.197
Connecting to cdn.pisugar.com (cdn.pisugar.com)|47.246.23.197|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 319036 (312K) [application/x-debian-package]
Saving to: ‘/tmp/pisugar-poweroff_1.6.1_armhf.deb’

/tmp/pisugar-poweroff_1.6.1_armhf.deb                    100%[================================================================================================================================>] 311.56K   487KB/s    in 0.6s

2021-11-08 08:12:35 (487 KB/s) - ‘/tmp/pisugar-poweroff_1.6.1_armhf.deb’ saved [319036/319036]

--2021-11-08 08:12:35--  http://cdn.pisugar.com/release/pisugar-programmer_1.6.1_armhf.deb
Resolving cdn.pisugar.com (cdn.pisugar.com)... 47.246.23.197
Connecting to cdn.pisugar.com (cdn.pisugar.com)|47.246.23.197|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 316464 (309K) [application/x-debian-package]
Saving to: ‘/tmp/pisugar-programmer_1.6.1_armhf.deb’

/tmp/pisugar-programmer_1.6.1_armhf.deb                  100%[================================================================================================================================>] 309.05K   630KB/s    in 0.5s

2021-11-08 08:12:37 (630 KB/s) - ‘/tmp/pisugar-programmer_1.6.1_armhf.deb’ saved [316464/316464]

Open I2C Interface
Uninstall old packages if installed
dpkg: warning: ignoring request to remove pisugar-server which isn't installed
dpkg: warning: ignoring request to remove pisugar-poweroff which isn't installed
dpkg: warning: ignoring request to remove pisugar-programmer which isn't installed
Install packages
/tmp/pisugar-server_1.6.1_armhf.deb /tmp/pisugar-poweroff_1.6.1_armhf.deb /tmp/pisugar-programmer_1.6.1_armhf.deb
Selecting previously unselected package pisugar-server.
(Reading database ... 45724 files and directories currently installed.)
Preparing to unpack .../pisugar-server_1.6.1_armhf.deb ...
Unpacking pisugar-server (1.6.1) ...
Selecting previously unselected package pisugar-poweroff.
Preparing to unpack .../pisugar-poweroff_1.6.1_armhf.deb ...
Unpacking pisugar-poweroff (1.6.1) ...
Selecting previously unselected package pisugar-programmer.
Preparing to unpack .../pisugar-programmer_1.6.1_armhf.deb ...
Unpacking pisugar-programmer (1.6.1) ...
Setting up pisugar-server (1.6.1) ...
Setting up pisugar-poweroff (1.6.1) ...
Setting up pisugar-programmer (1.6.1) ...
Clean up
Now navigate to http://192.168.1.220:8421 on your browser to see PiSugar power management
If you have any question,please feel free to contact us.
The PiSugar Team https://www.pisugar.com
pi-star@pi-star(rw):~$ sudo systemctl status pisugar-server
● pisugar-server.service - pisugar-server
   Loaded: loaded (/lib/systemd/system/pisugar-server.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2021-11-08 08:13:18 EST; 3min 50s ago
 Main PID: 5113 (pisugar-server)
    Tasks: 3 (limit: 991)
   CGroup: /system.slice/pisugar-server.service
[... 84 lines cut ...]
Nov 08 08:09:14 pi-star systemd[1]: pisugar-server.service: Failed with result 'exit-code'.
Nov 08 08:09:24 pi-star systemd[1]: pisugar-server.service: Service RestartSec=10s expired, scheduling restart.
Nov 08 08:09:24 pi-star systemd[1]: pisugar-server.service: Scheduled restart job, restart counter is at 7.
Nov 08 08:09:24 pi-star systemd[1]: Stopped pisugar-server.
Nov 08 08:09:24 pi-star systemd[1]: Started pisugar-server.
Nov 08 08:09:25 pi-star pisugar-server[1713]: [2021-11-08T13:09:25Z INFO  pisugar_server] TCP listening...
Nov 08 08:09:25 pi-star pisugar-server[1713]: [2021-11-08T13:09:25Z INFO  pisugar_server] WS listening...
Nov 08 08:09:25 pi-star pisugar-server[1713]: [2021-11-08T13:09:25Z INFO  pisugar_server] UDS listening...
Nov 08 08:09:25 pi-star pisugar-server[1713]: [2021-11-08T13:09:25Z INFO  pisugar_server] Http web server listening...
Nov 08 08:09:29 pi-star systemd[1]: Stopping pisugar-server...
Nov 08 08:09:29 pi-star systemd[1]: pisugar-server.service: Main process exited, code=killed, status=15/TERM
Nov 08 08:09:29 pi-star systemd[1]: pisugar-server.service: Succeeded.
Nov 08 08:09:29 pi-star systemd[1]: Stopped pisugar-server.
Nov 08 08:13:18 pi-star systemd[1]: Started pisugar-server.
Nov 08 08:13:18 pi-star pisugar-server[5113]: [2021-11-08T13:13:18Z INFO  pisugar_server] TCP listening...
Nov 08 08:13:18 pi-star pisugar-server[5113]: [2021-11-08T13:13:18Z INFO  pisugar_server] WS listening...
[... cut ...]
Nov 08 08:13:18 pi-star pisugar-server[5113]: [2021-11-08T13:13:18Z INFO  pisugar_server] Http web server listening...
Nov 08 08:13:18 pi-star pisugar-server[5113]: [2021-11-08T13:13:18Z INFO  pisugar_core] Init battery...
Nov 08 08:13:18 pi-star pisugar-server[5113]: [2021-11-08T13:13:18Z INFO  pisugar_core] Init rtc...
pi-star@pi-star(ro):~$ echo "get battery" | nc -q 0 /tmp/pisugar-server.sock
/tmp/pisugar-server.sock: forward host lookup failed: Unknown host
pi-star@pi-star(ro):~$ echo "get battery" | nc -u -q 0 127.0.0.1 8423
pi-star@pi-star(ro):~$
```

fengyc, 2021-11-09:
Try 1.6.2  https://github.com/PiSugar/pisugar-power-manager-rs/releases/download/latest/pisugar-server_1.6.2_armhf.deb

pchois, 2021-11-09:
> Try 1.6.2 https://github.com/PiSugar/pisugar-power-manager-rs/releases/download/latest/pisugar-server_1.6.2_armhf.deb

Same issue 

```
pi-star@pi-star(ro):~$ sudo systemctl status pisugar-server
● pisugar-server.service - pisugar-server
   Loaded: loaded (/lib/systemd/system/pisugar-server.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2021-11-09 08:23:01 EST; 2min 31s ago
 Main PID: 222 (pisugar-server)
    Tasks: 3 (limit: 991)
   CGroup: /system.slice/pisugar-server.service
           └─222 /usr/bin/pisugar-server --config config.json --model PiSugar 2 (2-LEDs) --web /usr/share/pisugar-server/web --http 0.0.0.0:8421 --ws 0.0.0.0:8422 --tcp 0.0.0.0:8423 --uds /tmp/pisugar-server.sock

Nov 09 08:25:32 pi-star pisugar-server[222]: [2021-11-09T13:25:32Z INFO  pisugar_core] Init battery...
Nov 09 08:25:32 pi-star pisugar-server[222]: [2021-11-09T13:25:32Z WARN  pisugar_server] Poll error: I/O error: Remote I/O error (os error 121)
Nov 09 08:25:32 pi-star pisugar-server[222]: [2021-11-09T13:25:32Z INFO  pisugar_core] Init battery...
Nov 09 08:25:32 pi-star pisugar-server[222]: [2021-11-09T13:25:32Z WARN  pisugar_server] Poll error: I/O error: Remote I/O error (os error 121)
Nov 09 08:25:32 pi-star pisugar-server[222]: [2021-11-09T13:25:32Z INFO  pisugar_core] Init battery...
Nov 09 08:25:32 pi-star pisugar-server[222]: [2021-11-09T13:25:32Z WARN  pisugar_server] Poll error: I/O error: Remote I/O error (os error 121)
Nov 09 08:25:32 pi-star pisugar-server[222]: [2021-11-09T13:25:32Z INFO  pisugar_core] Init battery...
Nov 09 08:25:32 pi-star pisugar-server[222]: [2021-11-09T13:25:32Z WARN  pisugar_server] Poll error: I/O error: Remote I/O error (os error 121)
Nov 09 08:25:32 pi-star pisugar-server[222]: [2021-11-09T13:25:32Z INFO  pisugar_core] Init battery...
Nov 09 08:25:32 pi-star pisugar-server[222]: [2021-11-09T13:25:32Z WARN  pisugar_server] Poll error: I/O error: Remote I/O error (os error 121)
pi-star@pi-star(ro):~$ i2cdump -y 1 0x75
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: XX 00 00 XX 00 XX 07 XX 00 XX XX 00 XX XX 00 XX    X..X.X?X.XX.XX.X
10: XX XX XX 00 1f XX 00 01 XX XX 00 XX XX 00 3f XX    XXX.?X.?XX.XX.?X
20: XX 00 XX 00 XX XX 00 00 XX XX 00 XX 00 XX XX 00    X.X.XX..XX.X.XX.
30: ff XX XX 00 XX XX 00 XX XX XX 00 XX XX 00 XX XX    .XX.XX.XXX.XX.XX
40: XX 00 XX XX XX 00 XX XX 07 XX XX 00 XX XX 00 XX    X.XXX.XX?XX.XX.X
50: XX 00 XX XX 00 XX XX XX 00 XX XX XX 00 XX XX XX    X.XX.XXX.XXX.XXX
60: 00 XX XX XX 00 XX XX 00 00 XX XX XX 00 07 XX XX    .XXX.XX..XXX.?XX
70: 00 XX XX XX 00 XX XX 00 00 XX XX XX 00 00 XX XX    .XXX.XX..XXX..XX
80: 00 0f XX XX 00 7f XX XX 00 XX 00 XX XX 00 00 XX    .?XX.?XX.X.XX..X
90: XX 00 XX XX 00 XX XX 00 00 XX XX XX 00 XX XX 00    X.XX.XX..XXX.XX.
a0: XX 00 ff XX 00 XX 00 00 XX XX 00 00 XX 00 XX XX    X..X.X..XX..X.XX
b0: 00 01 XX 00 XX XX 00 0f XX XX 00 XX XX XX 00 XX    .?X.XX.?XX.XXX.X
c0: 00 00 XX XX 00 00 XX XX 00 XX XX XX 00 XX XX XX    ..XX..XX.XXX.XXX
d0: 00 XX XX XX XX XX 00 ff XX XX 00 XX XX XX 00 XX    .XXXXX..XX.XXX.X
e0: XX XX XX XX XX XX 00 00 XX XX 00 00 XX XX XX XX    XXXXXX..XX..XXXX
f0: XX XX 00 XX XX XX 00 XX XX 00 XX XX XX 00 XX XX    XX.XXX.XX.XXX.XX
```

closed 2024-09-10 by github-actions[bot]