PiSugar/PiSugar #13: Strange colors when using with HyperPixel display

jzting, 2020-07-30:
I'm playing video with omxplayer on a 4" square HyperPixel display with a PiSugar2 and am seeing color issues similar to [this](https://github.com/pimoroni/fanshim-python/issues/70). Does the PiSugar interfere with the color data pins?

JdaieLin, 2020-07-30:
HyperPixel and PiSugar2 did use the same I2C pins. I2C bus is able to carry multiple devices.
Since it can still render correct image, the color issue is more likely caused by software.

jzting, 2020-07-30:
The color issue happens outside omxplayer (e.g. the colors are not showing correctly on the boot screen logo). When I remove the PiSugar, the colors return to normal.

Tobiah10, 2023-01-22:
TLDR: yes, the I2C pins that the PiSugar uses to communicate with the raspberry pi, are causing the color issues with the HyperPixel Display. 

The HyperPixel Displays use every GPIO pin to transfer data to the display including the I2C pins that are used to communicate to the PiSugar battery pack, but they are not used like normal I2C pins. Those pins are used more like general data pins for the display and when the pi is trying to talk with the PiSugar, those signals are causing your issues with the display. I currently have a project using a PiSugar 3 pack and a HyperPixel4 display and haven't had any issues yet, but I also haven't set up the PiSugar software as the HyperPixel guide explicitly says to not use any of the GPIO pins or digital interfaces because all the pins are used. The HyperPixel 4  display does break out a few pins that can be changed to function as an I2C Interface, and the install guide does explain how to change the I2C interface in the firmware, but the pogo pins on the PiSugar do not connect to them. I'm going to attempt to run a few jumper wires from the I2C port on the HyperPixel display to the I2C pins on the PiSugar and see if doing that will allow me to set up and use the PiSugar software.

benlk, 2023-08-09:
@Tobiah10 Did you have any luck running jumper wires? What do your connections look like?

Tobiah10, 2023-08-09:
@benlk 
Unfortunately, no. I tried a few different guides on setting up a secondary I2C bus on those pins but never got the PiSugar battery pack to communicate over it reliability. It worked now and again but, would randomly crash the webserver and freeze the entire pi sometimes. I eventually switched to an HDMI screen from waveshare and everything worked as intended. The main reason for my switch to a different screen was due to my project involving the use of a Camera Modual. When the OS sets up the resources for the camera modual, it reverts all I2C busses back to normal and locked them. The camera uses the I2C to send and receive some data. This threw everything I had done out the door. As far as wiring, I used the I2C breakout on the back of the HyperPixel display (my model came with a small plug installed, some don't apparently) and wired the corresponding SDA and SCL wires to the breakout pins on the back of the PiSugar battery pack. I did have to make a slight modification to the raspberry pi in order to get this to work and this may be where my Intermittent issues cane from. I had to break the normal I2C connection from the Raspberry pi to the PiSugar pack through the pogo-pins, otherwise there would be issues with signaling. I cut the pins as flush as I could to the board surface and coated them in nail polish to break that connection. That way I didn't destroy the pogo-pins on the PiSugar and it would work as intended if I ever reused it, and I could always remove the nail polish from the board with remover or heat if I needed that connection again.  I hope this helped and answered your question. I do belive it is possible to get working, I just wasn't completely successful.

closed 2024-09-09 by github-actions[bot]