PiSugar/PiSugar #14: PiSugar 2 hourly schedule instead of daily

guillaumedavidphd, 2020-09-29:
Is there a way to wake up the Pi on an hourly schedule? Or every 15 minutes for example? In the power management interface I only see a daily schedule.

JdaieLin, 2020-09-29:
In fact, the RTC chip supports this mode, but we haven't integrated it in the api yet.

guillaumedavidphd, 2020-09-29:
Got it. Is it on the roadmap at all?

JdaieLin, 2020-10-08:
We are considering this recently. Would you mind sharing with us in what scene you are going to use hourly schedule?

guillaumedavidphd, 2020-10-08:
Ideally I would love to be able to choose the granularity of the schedule. For example, it would allow me to acquire 5 minutes worth of data from sensors connected to the Pi every 20 minutes. In that scenario the Pi would be ON for 5 minutes and OFF for 15 minutes.

JdaieLin, 2020-10-09:
I see. The circle alarm mode would support this scene. Before we update this api, you can also schedule the next boot before shutting down.

JdaieLin, 2020-10-10:
You can use this script: https://github.com/PiSugar/pisugar-power-manager-rs/blob/master/scripts/power-on-off.sh

GSVNoFixedabode, 2021-01-01:
Would this script work for the Pi Sugar2 for a Pi Zero as well? I'm setting up a remote camera system (battery/solar) that needs to take a photo hourly during the day, so power on for 5 minutes then power off - every hour.

JdaieLin, 2021-03-06:
> Would this script work for the Pi Sugar2 for a Pi Zero as well? I'm setting up a remote camera system (battery/solar) that needs to take a photo hourly during the day, so power on for 5 minutes then power off - every hour.

Yes, it works for both PiSugar 2 and 2 pro.

seedrickMCWW, 2022-09-26:
This script stopped working for me. Did anything change?

jchain, 2022-12-27:
@JdaieLin Do you have any news on this power-on-off scheduling api? I feel like it is a much-needed feature.

closed 2024-09-09 by github-actions[bot]