PiSugar/PiSugar #15: Kernel support for SD3078

rrooggiieerr, 2020-10-07:
The default Raspberry Pi OS kernel currently doesn't support the SD3078 RTC, the driver is not included in the kernel.

PiSugar seems to "solve" this by talking directly to the i2c bus and basically reimplementing the whole driver in user space.

This way the common Linux commands `hwclock` and `timedatectl` as well as the `/sys/class/rtc/*` directory structure don't work.

I don't understand the design decision to use an RTC in PiSugar which is not supported by the Raspberry Pi OS, but it is what it is, so whatever.

However there is a kernel driver available, it only needs to be included in the Raspberry Pi OS kernel!

Just a simple request to the kernel maintainers was needed to have the driver included in future kernels, see https://github.com/raspberrypi/linux/issues/3881

So now comes an other problem, the SD3078 kernel driver is very basic and doesn't support a lot of features. Waking up at a certain time is not supported by the driver. The very common DS3231 RTC does implement these features in the kernel driver, and the driver is included in the default Raspberry Pi OS kernel. Again I don't understand the design decision to use the SD3078 in PiSugar.

Instead of implementing the features of the SD3078 in some Web UI, I think the existing kernel module should be extended so common Linux commands can be used to interact with the RTC. Using a Web UI to interact with hardware at a low level is in my opinion not the "Linux way" of doing things. Having a Web UI that interacts with common Linux commands and the /sys FS is a more proper way to implement this.

I'm interested in your thoughts on the above

fengyc, 2020-10-10:
@rrooggiieerr Yeah, you're right.  

SD3232 might be a better choice. At the time we started the project, we didn't have a sd3232 supplier, and sd3078 driver was not inclued in the raspberrypi kernel default configuration, we could not expected that anyone has the ability to compile the kernel all by oneself.

For these reasons, we provide a web ui and a text base api to make it easy to use. We are keeping updating hardware and software, and might replace the battery and rtc chip in the next release.

rrooggiieerr, 2020-10-12:
@fengyc Compiling a new kernel is actually very straightforward and documented on the Raspberry Pi website: https://www.raspberrypi.org/documentation/linux/kernel/building.md

Sure, choosing an other RTC in a _future_ version could be considered, but doesn't solve the lack of support for the RTC in the _current_ version of PiSugar. And on paper the SD3078 had some nice features, they're just not yet available in the kernel driver. I'm talking about the schedule functionality.

I think improving the kernel driver for the SD3078 is required to make the _current_ version of PiSugar a more finalised product. You already implemented the schedule functionality in your own software, so you know about the workings and which 12c registers need to be talked to. It "just" needs to be implemented in the kernel driver.

@zoroLi is the developer of the SD3078 kernel module, and @alexandrebelloni also did some work on it. Maybe you can persuade them to implement the missing functionality in the kernel?

alexandrebelloni, 2020-10-12:
Indeed, I'm the kernel maintainer for the RTC subsystem. I'm available for hire through Bootlin if you want me to improve the SD3078 driver. I there a public datasheet, from a cursory look, I couldn't find any? Else, I can definitively wrok from your existing code.

zoroLi, 2020-10-13:
@fengyc @rrooggiieerr 
I will improve the driver of sd3078 recently.

rrooggiieerr, 2020-10-13:
I love how this is coming together!

bgilreath, 2022-02-07:
Is there any update to this?

rrooggiieerr, 2022-08-11:
It doesn't look like anything has happened in the meantime, @zoroLi have you made any progress with improving the driver?

closed 2024-09-09 by github-actions[bot]