PiSugar/PiSugar #16: Kernel support for battery and button

rrooggiieerr, 2020-10-13:
I already created an issue for kernel support for the RTC, and that looks to be moving in the right direction, so I thought I also throw in my thoughts on the battery manager and button.

The Linux kernel supports a battery subsystem and also a key press subsystem

Why would you want this to be solved in the kernel?
* Create UDEV rules for battery levels and key presses
* Interact with battery trough `/proc/acpi/battery/*` directory structure
* Already existing battery tools will work with PiSugar, think about battery status icon for your X Windows environment, 

Here you can find information about writing kernel drivers for I2C devices:
https://www.kernel.org/doc/Documentation/i2c/writing-clients

Info on the Linux kernel Power Supply class:
https://www.kernel.org/doc/Documentation/power/power_supply_class.txt

I'm interested in your thoughts on the above

rrooggiieerr, 2024-08-23:
30 days? Almost 4 years!

rrooggiieerr, 2024-09-24:
Still an issue

JdaieLin, 2024-10-07:
PiSugar3 has kernel driver support. For more details, please refer to
https://github.com/PiSugar/pisugar-power-manager-rs/blob/74de74a0449fded35f080b9ab5cfdee076facb2a/pisugar-module/pisugar-3

As for PiSugar2, due to some communication issues with I2C, it doesn't support the kernel yet.

rrooggiieerr, 2024-10-07:
I've got the PiSugar2. What is the I2C issue, if it works using a script it should also be able to work using a kernel module I would assume

JdaieLin, 2024-10-09:
You're right, the I2C communication shouldn't affect the implementation of the kernel module. I'm discussing this with my colleagues and will work on implementing kernel support for PiSugar2 as soon as possible.

JdaieLin, 2024-11-15:
Kernel module for PiSugar2 is available here:
https://github.com/PiSugar/pisugar-power-manager-rs/tree/master/pisugar-module/pisugar-2

closed 2024-12-31 by github-actions[bot]