PiSugar/PiSugar #17: Interaction with other I2C devices

guillaumedavidphd, 2020-10-29:
Hi there,
I've noticed a consistent problem with PiSugar2 when used with other i2C devices. I'm currently trying to use a Pi Zero to log some environmental data and it has two I2C devices connected to it to that end. When I add the PiSugar2 and power the Pi with it the whole thing becomes unstable and the Pi completely loses power after a couple minutes. My devices are using I2C addresses that are different than the two I2C addresses used by the PiSugar2. I'd like to know if this is a known problem and if anyone else is experiencing this. I have 2 PiSugar2s connected to 2 different Pis and the same problem is occurring on both. This renders my PiSugar2s essentially useless for the original purpose I intended them for.

JdaieLin, 2020-11-11:
Hi, will the PiSuagr2 acts normally if you unload the extra I2C device?

guillaumedavidphd, 2020-11-11:
Both of my PiSugar2s work correctly without the I2C devices attached.

JdaieLin, 2020-11-22:
@wlmh110

wlmh110, 2020-11-23:
Hi,
Could you please provide what data is in I2C when all the devices work together?
And what devices do you use? If we can get one, it will more easily solve this problem.

guillaumedavidphd, 2020-11-24:
Could you specify what you mean by "provide what data is in I2C"? Here are the 2 I2C devices that I'm using in concurrence with PiSugar2:

- https://www.adafruit.com/product/4632
- https://www.adafruit.com/product/4636

Thanks for looking into it!

wlmh110, 2020-11-25:
Please use the i2c tool to read the address of 0x75 and 0x32.
i2cdump -y 1 0x32
i2cdump -y 1 0x75
Then you can see the data in these two addresses.

guillaumedavidphd, 2020-12-02:
It's a bit tricky. Since the PiSugar2 was not working, I removed it from my design and currently everything is packaged and in use. It would be a lot faster if you got your hands on the sensors I mentioned I tried it out for yourself.

tonyrog, 2020-12-10:
Here is an output where 0x32 is up and may be "talked" to, but 0x75 is gone.
Had this problem before but then 0x75 came back... (piSugar2)

i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- 32 -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

tonyrog, 2020-12-10:
I turned piSuger2 off and pressed reset button for a while, then 0x75 came back.
Had not tried that sequence before...

i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- 32 -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- 75 -- --

I turn piSugar2 on and then after a while it is gone again...

Repeated off - press reset - and it show up again, for a while then gone
even in off mode.

tonyrog, 2020-12-10:
If I repeat the procedure I can read voltage and initialize the ip5209 module (while in off mode) but after about 10-20 seconds it is gone again.  A press on reset will make it appear again and voltage may be read. But then ... all of this in off mode. Please? any way out of this?

wlmh110, 2020-12-13:
> If I repeat the procedure I can read voltage and initialize the ip5209 module (while in off mode) but after about 10-20 seconds it is gone again. A press on reset will make it appear again and voltage may be read. But then ... all of this in off mode. Please? any way out of this?

If 0x32 is working well, then the reason 0x75 doesn't show up because one other power supply is connected to the Raspberry Pi, please see issue #18

fabiantechera, 2021-06-25:
Hello! I have the same problem with PiSugar 2 Pro and a BME280 sensor. I do not get data from the BME sensor and also PiSugar doesn`t show information about charging and anything else.
Some help to get both works independently?
Thank you.

Some additional information:

It supposed that BME280 uses the 0x76 address

While Pisugar and BME280 are connected I get this from the console:

pi@raspberrypi:~ $ sudo i2cdump -y 1 0x32
No size specified (using byte-data access)
0 1 2 3 4 5 6 7 8 9 a b c d e f 0123456789abcdef
00: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XXXXXXXXXXXXXXXX
10: XX XX XX XX XX XX ^C
pi@raspberrypi:~ $ sudo i2cdump -y 1 0x75
No size specified (using byte-data access)
0 1 2 3 4 5 6 7 8 9 a b c d e f 0123456789abcdef
00: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX ^C
pi@raspberrypi:~ $ sudo i2cdump -y 1 0x75
No size specified (using byte-data access)
0 1 2 3 4 5 6 7 8 9 a b c d e f 0123456789abcdef
00: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XXXXXXXXXXXXXXXX
10: XX XX XX XX

pi@raspberrypi:~ $ sudo i2cdump -y 1 0x76
No size specified (using byte-data access)
0 1 2 3 4 5 6 7 8 9 a b c d e f 0123456789abcdef
00: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XXXXXXXXXXXXXXXX

When BME280 not connected, only PiSugar Connected:

pi@raspberrypi:~ $ sudo i2cdump -y 1 0x32
No size specified (using byte-data access)
0 1 2 3 4 5 6 7 8 9 a b c d e f 0123456789abcdef
00: 12 01 a1 04 01 04 61 00 00 00 00 00 00 00 00 00 ??????a.........
10: 10 00 00 00 00 00 20 00 00 00 e0 0f 00 00 08 39 ?..... ...??..?9
20: 11 97 06 20 03 61 38 81 04 18 03 61 51 cf 79 fc ??? ?a8????aQ?y?
30: cd f2 0a b7 9f 71 df 22 b5 6a 02 35 bb 49 7d df ?????q?"?j?5?I}?
40: bd 7f bd b9 6d 35 5f 17 4f 56 6e 57 40 6f f3 f7 ????m5_?OVnW@o??
50: e6 d6 ff db c3 82 20 ce d5 54 ad 78 9a cf f3 c9 ??.??? ??T?x????
60: 6e cb 7b 76 fd f9 ec df 75 1d e1 9d 1c 7e 3e e6 n?{v????u????~>?
70: 76 3b 20 11 25 32 20 44 10 34 ff ff ff ff ff ff v; ?%2 D?4......
80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ................
90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ................
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ................
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ................
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ................
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ................
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ................
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ................

pi@raspberrypi:~ $ sudo i2cdump -y 1 0x75
No size specified (using byte-data access)
0 1 2 3 4 5 6 7 8 9 a b c d e f 0123456789abcdef
00: ea ff 77 26 bf 00 00 87 03 d0 25 00 fd 1e 04 03 ?.w&?..???%.????
10: 65 df 76 d5 0b 61 03 00 a0 be 00 b8 f8 20 00 00 e?v??a?.??.?? ..
20: ff 8a 92 08 d4 60 a6 63 4a b7 df 8f 28 00 00 00 .????`?cJ???(...
30: 3f 26 1a 70 d8 89 c3 c0 ba dc 11 40 d8 02 ad c0 ?&?p???????@????
40: 19 3d 11 57 55 be d4 e9 00 00 00 00 00 00 00 00 ?=?WU???........
50: 05 56 46 71 02 00 04 00 00 00 00 80 00 00 00 00 ?VFq?.?....?....
60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
70: 37 ba 00 00 08 82 c0 58 00 00 00 26 03 13 08 48 7?..???X...&???H
80: 7f 8c a6 d0 5a 76 b6 87 b0 7f 8c a6 d0 5a 76 b6 ????Zv???????Zv?
90: 87 b0 53 66 86 af 56 86 be 5f 91 53 66 86 af 56 ??Sf??V???Sf??V
a0: 86 be 5f 91 b5 3a e9 7e 00 00 00 00 00 00 00 00 ????:?........
b0: f0 93 93 b3 51 7e 00 00 00 00 00 00 00 00 00 00 ????Q..........
c0: 8d 55 f7 fa 63 23 06 f6 eb fa 00 00 00 00 00 00 ?U??c#????......
d0: ba 0d cb 3e 9a 26 e2 01 00 00 00 00 c2 38 7a 0e ???>?&??....?8z?
e0: c0 00 00 81 1b 00 00 53 09 00 00 00 11 8d 10 00 ?..??..S?...???.
f0: 08 43 02 61 0a 00 00 70 01 0b 10 b9 00 00 00 00 ?C?a?..p????....

pi@raspberrypi:~ $ sudo i2cdump -y 1 0x76
No size specified (using byte-data access)
0 1 2 3 4 5 6 7 8 9 a b c d e f 0123456789abcdef
00: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XXXXXXXXXXXXXXXX
10: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XXXXXXXXXXXXXXXX
20: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XXXXXXXXXXXXXXXX
30: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XXXXXXXXXXXXXXXX
40: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XXXXXXXXXXXXXXXX
50: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XXXXXXXXXXXXXXXX
60: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XXXXXXXXXXXXXXXX
70: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XXXXXXXXXXXXXXXX
80: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XXXXXXXXXXXXXXXX
90: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XXXXXXXXXXXXXXXX
a0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XXXXXXXXXXXXXXXX
b0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XXXXXXXXXXXXXXXX
c0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XXXXXXXXXXXXXXXX
d0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XXXXXXXXXXXXXXXX
e0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XXXXXXXXXXXXXXXX
f0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XXXXXXXXXXXXXXXX

jan94, 2021-07-07:
I had the same issue and ended up creating a second i2c bus...

fabiantechera, 2021-07-07:
Hello jan94, thanks for your answer. 
Yes, I did the same. Everything ok now. 
Regards

tamugo, 2021-07-23:
Hello @fabiantechera I have the same issue with not being able to detect the PiSugar address when another device is connected. Can you tell me how you created the bus you used?

fabiantechera, 2021-07-26:
Hello @tamugo. I followed this guide.
https://www.instructables.com/Raspberry-PI-Multiple-I2c-Devices/

mlaity, 2021-11-10:
I am experiencing the same issue, using a Raspberry Pi 4B, OzzMaker BerryGPS-IMU-4, and PiSugar 2 Pro.  Were you ever able to resolve this other than creating a new i2c bus?  We can't just create additional I2C buses as both devices have built-in headers to connect to I2C Bus 1 on Pi Pins 3 and 5.

The I2C addresses continually come and go for each device, making both devices inaccessible.  There does not seem to be a conflict in I2C addresses between these devices, with the BerryGPS-IMU-v4 using 1c, 6a, and 77, and the PiSugar using 32 and 75.

fabiantechera, 2021-11-10:
Hello mlaity,
I know what you mean. Creating a new i2c bus works for me. I didn`t try any other solution because y made the connection with cables in different pins. Regards!

closed 2024-09-01 by github-actions[bot]