PiSugar/PiSugar #18: ERROR pisugar_server] Invalid slave address: 117

sinoia, 2020-10-31:
● pisugar-server.service - pisugar-server
   Loaded: loaded (/lib/systemd/system/pisugar-server.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2020-10-31 07:42:07 GMT; 19min ago
 Main PID: 267 (pisugar-server)
   Memory: 4.7M
   CGroup: /system.slice/pisugar-server.service
           └─267 /usr/bin/pisugar-server --config config.json --model PiSugar 2 (2-LEDs) --web /usr/share/pisugar-server/web --ht

Oct 31 08:01:27 raspberrypi pisugar-server[267]: [2020-10-31T08:01:27Z ERROR pisugar_server] Invalid slave address: 117
Oct 31 08:01:27 raspberrypi pisugar-server[267]: [2020-10-31T08:01:27Z ERROR pisugar_server] Invalid slave address: 117
Oct 31 08:01:28 raspberrypi pisugar-server[267]: [2020-10-31T08:01:28Z ERROR pisugar_server] Invalid slave address: 117
Oct 31 08:01:28 raspberrypi pisugar-server[267]: [2020-10-31T08:01:28Z ERROR pisugar_server] Invalid slave address: 117
Oct 31 08:01:28 raspberrypi pisugar-server[267]: [2020-10-31T08:01:28Z ERROR pisugar_server] Invalid slave address: 117
Oct 31 08:01:29 raspberrypi pisugar-server[267]: [2020-10-31T08:01:29Z ERROR pisugar_server] Invalid slave address: 117
Oct 31 08:01:29 raspberrypi pisugar-server[267]: [2020-10-31T08:01:29Z ERROR pisugar_server] Invalid slave address: 117
Oct 31 08:01:30 raspberrypi pisugar-server[267]: [2020-10-31T08:01:30Z ERROR pisugar_server] Invalid slave address: 117
Oct 31 08:01:30 raspberrypi pisugar-server[267]: [2020-10-31T08:01:30Z ERROR pisugar_server] Invalid slave address: 117
Oct 31 08:01:30 raspberrypi pisugar-server[267]: [2020-10-31T08:01:30Z ERROR pisugar_server] Invalid slave address: 117

JdaieLin, 2020-11-11:
@fengyc

fengyc, 2020-11-12:
@sinoia Whether the i2c function is enabled or not? See https://github.com/PiSugar/pisugar-power-manager-rs#prerequisites

timgunter, 2020-11-23:
I am also having this issue with a new pisugar2 2-led model. Pisugar-server is able to connect occasionally, but most of the time it cannot and the "invalid slave address: 117" error shows up in the system log. I have another I2C device at address 0x1a that is working.

tonyrog, 2020-12-10:
I have the same problem on 2 devices the 117 (0x75) slave does not show up.
RTC 50 (0x32) do show up. They started behaving strange after a while. Any workarounds? I have 2 more that are not in use yet!

wlmh110, 2020-12-13:
> I have the same problem on 2 devices the 117 (0x75) slave does not show up.
> RTC 50 (0x32) do show up. They started behaving strange after a while. Any workarounds? I have 2 more that are not in use yet!

The reason 0x75 doesn't show up because one other power supply is connected to the Raspberry Pi.
If you use PiSuga2 on Pwnagotchi, we suggest you use this image, we have installed PiSugar Plugin that can show the battery level on the E-ink screen:
https://mega.nz/file/emwmjZhZ#GKE_yJ9i14KB6S9o3kcSokmQy-FacAaNd2b2GQd-EZY

wlmh110, 2020-12-13:
> I am also having this issue with a new pisugar2 2-led model. Pisugar-server is able to connect occasionally, but most of the time it cannot and the "invalid slave address: 117" error shows up in the system log. I have another I2C device at address 0x1a that is working.

If 0x32 is working well, then the reason 0x75 doesn't show up because one other power supply is connected to the Raspberry Pi.

tonyrog, 2020-12-13:
Why would that be a problem? And how can that explain the things we are seeing? Is it a bug or is it designed that way? If it is a bug, can we mod the product some how to make it accept another USB cable? In that application me and my colleague is working on we are using the gadget port with USB g_ether. Why does the i2c devices show up for 10-20 seconds when I push the tap button? I am not even using the button for anything.

3dlabsio, 2020-12-27:
Also experiencing this issue on a Compute Module 4 with CM4 IO Board. Is there a way to change the occupied i2c address? We are integrating the pisugar into a project and if this doesn't work we'll have to return all the units we purchased.

wlmh110, 2021-01-02:
> Why would that be a problem? And how can that explain the things we are seeing? Is it a bug or is it designed that way? If it is a bug, can we mod the product some how to make it accept another USB [...]

Hi, It designed like this. When an extra power source is connected, Pi will get power from the extra power source, then PiSugar has nothing to do, so it will shut down automatically.
We add an option in the 1.4.8 version, it will close the automatic shutdown function, so even there is an extra power source, PiSugar won't shutdown.
But be careful, while this option is chosen, Pisugar can only be shut down with the system when using the shutdown command or the battery is run out. Even turn the switch off, the power IC will still work.

wlmh110, 2021-01-02:
> Also experiencing this issue on a Compute Module 4 with CM4 IO Board. Is there a way to change the occupied i2c address? We are integrating the pisugar into a project and if this doesn't work we'll have to return all the units we purchased.
> 
> The addresses appear to be detected:
> $ i2cdetect -y 1
> 0 1 2 3 4 5 6 7 8 9 a b c d e f
> 00: -- -- -- -- -- -- -- -- -- -- -- -- --
> 10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 30: -- -- 32 -- -- -- -- -- -- -- -- -- -- -- -- --
> 40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 70: -- -- -- -- -- 75 -- --
> 
> This is the data in each address:
> $ i2cdump -y 1 0x32
> No size specified (using byte-data access)
> 0 1 2 3 4 5 6 7 8 9 a b c d e f 0123456789abcdef
> 00: 27 21 26 06 24 02 01 00 00 12 00 00 00 00 00 00 '!&?$??..?......
> 10: 00 00 00 00 00 00 1b 00 82 00 e0 38 00 00 fa 33 ......?.?.?8..?3
> 20: 29 01 00 04 02 01 02 21 04 22 02 01 56 32 2e 30 )?.????!?"??V2.0
> 30: 50 69 53 75 67 61 72 20 54 65 61 6d 20 20 20 20 PiSugar Team
> 40: 4d 61 64 65 20 62 79 20 68 75 6d 61 6e 20 6f 6e Made by human on
> 50: 65 61 72 74 68 2e 20 20 20 20 20 20 20 20 20 20 earth.
> 60: 77 77 77 2e 50 69 53 75 67 61 72 2e 63 6f 6d 20 [www.PiSugar.com](http://www.PiSugar.com)
> 70: 20 20 20 07 22 41 20 23 12 45 ff ff ff ff ff ff ?"A #?E......
> 80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ................
> 90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ................
> a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ................
> b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ................
> c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ................
> d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ................
> e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ................
> f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ................
> 
> $ i2cdump -y 1 0x75
> No size specified (using byte-data access)
> 0 1 2 3 4 5 6 7 8 9 a b c d e f 0123456789abcdef
> 00: ea ff 77 26 bf 00 80 87 03 d0 25 00 fd 1e 04 03 ?.w&?.????%.????
> 10: 65 df 76 e7 0b 61 03 00 a0 be 00 b8 f8 20 00 00 e?v??a?.??.?? ..
> 20: ff 8a 92 08 d4 60 a6 63 4a f7 df 8f 28 00 00 00 .????`?cJ???(...
> 30: 30 21 15 70 d4 89 c3 c0 ba dc 11 40 d8 02 ad c0 0!?p???????@????
> 40: 19 3d 11 57 55 be d1 e9 00 00 00 00 00 00 00 00 ?=?WU???........
> 50: 05 56 00 71 00 00 00 00 00 00 00 80 00 00 00 00 ?V.q.......?....
> 60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
> 70: 53 ba 00 00 00 82 c0 58 00 00 00 24 03 09 08 88 S?...??X...$????
> 80: 7f 8c a6 d0 5a 76 b6 87 b0 7f 8c a6 d0 5a 76 b6 ????Zv???????Zv?
> 90: 87 b0 53 66 86 af 56 86 be 5f 91 53 66 86 af 56 ??Sf??V??_?Sf??V a0: 86 be 5f 91 b5 3a e9 73 00 00 00 00 00 00 00 00 ??_??:?s........
> b0: f0 81 89 a8 46 73 00 00 00 00 00 00 00 00 00 00 ????Fs..........
> c0: 8d 55 e7 fa 63 23 06 f6 eb d0 00 00 00 00 00 00 ?U??c#????......
> d0: fb 0f e3 3e a3 26 ce 01 e6 39 00 00 00 00 70 10 ???>?&???9....p?
> e0: c0 00 00 81 1b 00 00 51 1c 00 00 00 11 0a 10 00 ?..??..Q?...???.
> f0: 08 43 02 61 0a 00 00 70 01 0b 10 b9 00 00 00 00 ?C?a?..p????....

We don't quite understand what problem you have met. It seems that both addresses are working well.
We can't change the I2C address of PiSugar, may think about this in the next generation.

tonyrog, 2021-01-02:
> Hi, It designed like this. When an extra power source is connected, Pi will get power from the extra power source, then PiSugar has nothing to do, so it will shut down automatically.
> We add an [...]

Ok. Thank you for the reply. I think I may need to use that option since we use a gadget ether to communicate sometimes. I will report back on the progress on this.

matthiasg, 2021-12-30:
What about address 50 ? What is that used for ?

fengyc, 2021-12-31:
> What about address 50 ? What is that used for ?

0x32 is the i2c address of rtc chip in pisugar2.

matthiasg, 2021-12-31:
That explains why I had rtc issues. Updated to 1.6.x and the problem vanished. Wasn't the soldering though but an issue with raspberry zero 2 compatibility

xlla, 2022-09-29:
> If you use PiSuga2 on Pwnagotchi, we suggest you use this image, we have installed PiSugar Plugin that can show the battery level on the E-ink screen:

hi @wlmh110 ,  how to show the battery level on the E-ink screen using my pwnagotchi build?

wlmh110, 2022-09-30:
> > If you use PiSuga2 on Pwnagotchi, we suggest you use this image, we have installed PiSugar Plugin that can show the battery level on the E-ink screen:
> 
> hi @wlmh110 , how to show the battery level on the E-ink screen using my pwnagotchi build?

Hi, please use this image, which we have installed all part already:
https://mega.nz/file/8XcUQZhD#TngZV9OZaPXLWnqIwmvzJeYdL0ddT33S21EQAiARcwc

closed 2024-08-31 by github-actions[bot]