PiSugar/PiSugar #19: Is there a reason I get low power output sometimes?

justanothermatt, 2020-12-21:
I'm using the PiSugar2 with a Raspberry Pi Zero and a a PiOLED. Sometimes when I use the battery the PiOLED only gets ~1.3V on pins 3 and 5. Because of this, my screen won't turn on. Is there any reason this would be happening?

JdaieLin, 2021-01-02:
@wlmh110

wlmh110, 2021-01-02:
> I'm using the PiSugar2 with a Raspberry Pi Zero and a a PiOLED. Sometimes when I use the battery the PiOLED only gets ~1.3V on pins 3 and 5. Because of this, my screen won't turn on. Is there any reason this would be happening?

Hi, we haven't seen this yet.
Could you describe under what circumstances this happened?

justanothermatt, 2021-01-04:
PiSugar 2 is connected to a Pi Zero W. Which has [this ](https://www.adafruit.com/product/3527) PiOLED on top of it. When I plug the Pi Zero into external power the screen works fine. When I use the battery sometimes it will work and other times it wont. The voltage on pins 3 and 5 are too low and `sudo i2cdetect -y 1` shows weird values typically looking something like this:

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 04 -  -   -   - 09 0a 0b 0c 0d -  -
10: -  -  - 13 14 15 16 - -  -  -  1b 1c 1d 1e -
20: - - - 23 24 25 26 27 - - -  - 2c 2d 2e 2f
30: 30 - -  -  - 35 36 37 - -  -  -  3c 3d 3e 3f
40: 40 - -  - - 45 46 47 48 49 - - - - 4e 4f
50: 50 51 - - -  -  -  57 58 - - -  -  -   - 5e 5f
60: 60 - - -  -  -  66 67 68 69 -  - -  - 6e 6f
70: 70 71 72 73 - - - -
```

Note that it doesn't ALWAYS look like this. Sometimes the occupied addresses are different. It seems to me like there's an issue during initial boot of the PiSugar. How does it interact with the i2c bus on boot? It seems like it's initializing every address but getting interrupted at some point and this issue is happening

wlmh110, 2021-01-05:
Hi,
It seems that PiSugar can’t work with PiOLED.
This is because Raspberry Pi use an nonstandard design on I2C, it pull down
the voltage instead of pull up.
PiSugar only can pull up the voltage of I2C a little.
So it can work with most of devices, but due to the IC design, when it been
pull down too low on the boot time, I2C interface will turn into LED
interface, it cause the mess.
We will fix this problem in the next version of PiSugar, but we can’t do
anything about it on PiSugar 2/PiSugar 2 Pro.
We are very sorry about that.
PiSugar Team

justanothermatt, 2021-01-05:
Ah I understand. I've seen photos of screen working with the PiSugar, would any of those be able to work with PiSugar 2? If so, could you link to those screens?

justanothermatt, 2021-01-07:
Been doing research on this topic the past couple of days and I can't find any documentation that says the Raspberry Pi pulls down I2C. In fact everything I see says they have [two pull up resistors on I2C](https://forums.adafruit.com/viewtopic.php?f=8&p=639992#:~:text=The%20RasPi%20and%20Arduino%20both,value%20(20k%20to%2040k).) Any ideas with that?

wlmh110, 2021-01-07:
Hi，
It's two different things.
We need to pull up the voltage before the system power-up, in another word,
there should be a large resistance between I2C interface and the GND.

You can test the voltage between the I2C interface and the GND while the
system is off.
If PiSugar connects nothing, normally it will be 3V.

We have used some screens that don't use the I2C interface, and they work
well. But as we know, for now, all adafruit screens do not work well with
PiSugar2.
Hope that helps
PiSugar Team

justanothermatt, 2021-01-07:
Do you have links to the SPI screen you have used?

wlmh110, 2021-01-10:
Hi,
Yes, we have tested some SPI screens, but we can't find the same name on
Amazon.
We tried to send you the picture, but the mail has been denied by the
server.
Here is the link to Taobao
https://item.taobao.com/item.htm?spm=a1z09.2.0.0.69382e8d2GvgHO&id=615754762052&_u=ahqg6qa8241
Hope that helps
PiSugar Team

closed 2021-01-11 by justanothermatt