PiSugar/PiSugar #20: Server doesn't seem to shut down the pi based on PiSugar2's battery or button?

mildmojo, 2020-12-22:
I just got a PiSugar2. I think it's the 4-LED version, though I only count 3 (blue charging, green full charge, blue power on). I used the installer script from the wiki to install the server software into a DIetPi installation from videolooper.de (image v2.3), running on an old Pi Zero v1.2. The server is running, I can visit the very pretty control page which displays the battery status, RTC info, and lets me change settings.

When I use the web UI to configure the PiSugar button to shut the pi down on a long press, it doesn't seem to have any effect. I can hold the button for 10 seconds or more and nothing happens. Same for short or double-press settings. I can see the output changing in `sudo i2cdetect -y 1 0x32` when I press the button. A bunch of 'X' characters appear on the ASCII output side.

To test the low-battery shutdown, I used the web UI to set it to shut down immediately at 3% battery, then edited `/etc/pisugar-server/config.json` and set "auto_shutdown_level" to "95.0" for testing. I restarted the server with `sudo systemctl restart pisugar-server`, visited the control page, and left it open until the battery percentage hit 95%. Nothing appeared in `dmesg` logs, and the pi did not shut down.

I also have an Adafruit 2.8" resistive-touch PiTFT LCD attached.

Any advice? Thanks!

_(Also: It's not clear to me if the UI is configuring the PiSugar firmware settings, the server settings, or both. It's also not clear to me what functions the server is responsible for, and what the PiSugar is responsible for. When setting the low-battery shutdown, does the server watch for that? Or does the PiSugar watch, and use I2C to signal the server to shut the system down? When configuring the delay for low-battery shutdown, is that a delay between the low battery condition and the `sudo shutdown now`, or a delay between the shutdown command and the PiSugar firmware poweroff? Do I have to configure anything manually to make the PiSugar allow the system to shut down before powering off, or is that how the system already behaves?)_

JdaieLin, 2021-01-02:
It should be 2-LED version. 4-led refers to 4 leds indicating power level, not all leds in total.

mildmojo, 2021-02-20:
Ah, thanks! I think this fixed the shutdown issues.

closed 2021-02-20 by mildmojo