PiSugar/PiSugar #21: Pisugar2 Pro / 5000mah+ PI4 not charging, stuck at 26% for 2 days...

Cocoo68332, 2020-12-23:
Hi,

I just installed a PisugarPro 2 + 5000mha battery on a RPI4 and it does not seem to charge the battery is at 26% since install. 

I pressed the swith "OFF" during the night, expecting the RPI4 to stop working and the battery to charge, but it is still stuck at 26%

I checked the battery charge with the web UI and the i2c cli interface as root.

Is this normal ?

JdaieLin, 2021-01-02:
@wlmh110

wlmh110, 2021-01-02:
> Hi,
>
> I just installed a PisugarPro 2 + 5000mha battery on a RPI4 and it does not seem to charge the battery is at 26% since install.
>
> I pressed the swith "OFF" during the night, expecting the RPI4 [...]

Hi, it's very abnormal.
Which port do you use for charging? While charging, have you saw the blue or green LED light up?

Cocoo68332, 2021-01-02:
I think I solved the problem by choosing the "pro" model in the config
wizard, I picked the wrong model at the beginning.

Thx

L

closed 2024-08-31 by github-actions[bot]