PiSugar/PiSugar #22: Problem installing power manager

beautifulmount, 2020-12-29:
When trying to install power manager i got this message "curl: (6) Could not resolve host: cdn.pisugar.com"

JdaieLin, 2021-01-02:
Hi, where is your location? We need to check the accessibility of the CDN.

JdaieLin, 2021-01-02:
We've change the region to global. You can try a again later.

beautifulmount, 2021-01-02:
Hi, still not working. I'm in Italy.

JdaieLin, 2021-01-09:
If it's still not accessible, you should download the pisugar-server an pisugar-poweroff from:
https://github.com/PiSugar/pisugar-power-manager-rs/releases/tag/v1.4.8

and install them using script:
dpkg -i xxx.deb

beautifulmount, 2021-03-17:
The dpkg seems to work fine but then appears "connection closed by the remote host" and since then the pi fail to boot

JdaieLin, 2021-04-10:
@fengyc

fengyc, 2021-04-10:
> The dpkg seems to work fine but then appears "connection closed by the remote host" and since then the pi fail to boot

@beautifulmount  Any screenshots or systemd log?

pcfreaxx, 2022-06-16:
I have the same Problem. I am in Germany but also with Hongkong vpn still curl could not resolve host cdn.pisugar.com
I send my pisugar2 back. Without working Software, it is just an expensive battery.

invictusani, 2022-09-17:
@JdaieLin I'm having install the software as well, im located in the US. I get the code curl: "(6) Could not resolve host: cdn.pisugar.com"

any advice?

pcfreaxx, 2022-09-17:
@invictusani this error is known. Go on GitHub and download each package file by Hand and Install via dpkg -i

invictusani, 2022-09-17:
@pcfreaxx dang alright, ill do that. LOL yeah im having a hard time getting this thing set up.

thanks again for the help!

randoassuser, 2024-04-19:
why exactly does it not work via curl? problem with your ssl certificates?
and how to install the packages via dpkg -i

closed 2024-08-30 by github-actions[bot]