PiSugar/PiSugar #23: Question - Solar Power

mikev1963, 2021-01-04:
Can I hook up a solar panel to charge battery while using the Pi Sugar Pro?  

Not sure where rot ask questions.

JdaieLin, 2021-01-09:
Yes you can. The solar panel should have its own output board, connect it to the 5V input pads.

JdaieLin, 2021-01-09:
'qi' VIN and GND

PEBL-CIC, 2023-10-12:
Hi, so you have to use a separate charge controller between the panel and the PiSugar? Do you recommend any boards for this use?

machs123, 2024-01-04:
I have a PiSugar 3 I would like to charge using a 5V solar panel.  This thread suggests such panels should be connected to the qi input of the PiSugar 3 board.  My question relates to the maximum input voltage the board allows.  I am considering a solar panel specified for 5V and 5W.  However, this panel has the following additional specifications - 6.13V peak and 7.13V open circuit.  Can the PiSugar 3 qi input handle these voltages?  I just don't want to damage my new PiSugar 3.  Alternately, I assume the PiSugar 3 uses a power management IC to manage the battery, AC adapter and qi inputs used for charging the battery.  If someone can tell me what IC is used I could look it up and share my experience with others having similar questions.  I really like my PiSugar 3 and look forward to someone helping me with this.

closed 2024-08-29 by github-actions[bot]

seschu, 2024-10-29:
> I have a PiSugar 3 I would like to charge using a 5V solar panel. This thread suggests such panels should be connected to the qi input of the PiSugar 3 board. My question relates to the maximum [...]

ever found a solution? @machs123

robertlagrant, 2026-09-22:
I'd also love to know if anyone's done this, and what panel they used.

funkymonkey5150, 2026-09-23:
Best regards,Steve PlasenciaBusiness Development ManagerCell: (323) 573-1967https://calendly.com/steve-plasencia/staffing_services

funkymonkey5150, 2026-09-23:
Best regards,Lm> wrote:robertlagrant left a comment (PiSugar/PiSugar#23)
I'd also love to know if anyone's done this, and what panel they used.