PiSugar/PiSugar #27: PiSuger issues with read only file systems

CMDRPhaedra, 2021-01-19:
I'm new to Pi's and my first project is to install a PiSuger device to act as a UPS on a device configure with TelsaUSB.. however the control software doesn't seem to run when the filesystem is read only. Its set as read only to stop the system becoming corrupt when the Tesla cuts the power.... "this is why I got.a Pisuger"

pi@teslausb:~ $ sudo systemctl status pisugar-server
● pisugar-server.service - pisugar-server
Loaded: loaded (/lib/systemd/system/pisugar-server.service; enabled; vendor preset: enabled)
Active: activating (auto-restart) (Result: exit-code) since Tue 2021-01-19 18:03:59 GMT; 3s ago
Process: 1110 ExecStart=/usr/bin/pisugar-server $OPTS (code=exited, status=1/FAILURE)
Process: 1115 ExecStopPost=/bin/rm -f /tmp/pisugar-server.sock (code=exited, status=0/SUCCESS)
Main PID: 1110 (code=exited, status=1/FAILURE)

when I make the filesystem read/write it starts working..

an 19 18:11:00 teslausb systemd[1]: Started pisugar-server.
Jan 19 18:11:00 teslausb pisugar-server[2513]: [2021-01-19T18:11:00Z INFO pisugar_server] TCP listening...
Jan 19 18:11:00 teslausb pisugar-server[2513]: [2021-01-19T18:11:00Z INFO pisugar_server] WS listening...
Jan 19 18:11:00 teslausb pisugar-server[2513]: [2021-01-19T18:11:00Z INFO pisugar_server] UDS listening...
Jan 19 18:11:00 teslausb pisugar-server[2513]: [2021-01-19T18:11:00Z INFO pisugar_server] Http web server listening...

any ideas?

JdaieLin, 2021-03-06:
@fengyc

fengyc, 2021-03-07:
@CMDRPhaedra Edit `/etc/default/pisugar-server`，remove `--uds xxxxx`

closed 2024-08-27 by github-actions[bot]