PiSugar/PiSugar #28: BerryGPS-IMU compatibility

mhgwilliams, 2021-01-28:
Hello,

I'm using the pisugar 2 with the raspberry pi zero w. I love the battery, super easy to get started and the form factor is really nice to use. However, I'm noticing some issue with my i2c devices when I use the battery. For whatever reason, when running 'i2cdetect -y 1' after installing both the pisugar 2 and the berrygps-imu hat, I get intermittent listing of devices. If I run that command over and over, sometimes I'll get correct listings, and other times the address where the device was is listed as empty. I checked my wiring and it doesn't seem to be the issue. when running via usb I don't get this issue. I double checked the addresses on the i2c bus aren't overlapping, so that doesn't seem to be the problem either.

[BerryGPS-IMU for reference.](https://ozzmaker.com/product/berrygps-imu/)

Am I missing something simple? not sure where this instability is coming from, would love a suggestion on how to troubleshoot if anyone else has similar issues.

mhgwilliams, 2021-01-28:
Saw in issue #17 you requested specific addresses info dumped, so here's that preemptively.

pi@pizero:~ $ i2cdump -y 1 0x32
> No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
10: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
etc, etc...

pi@pizero:~ $ i2cdump -y 1 0x75

> No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
10: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
etc, etc...

Here is what I get with all devices connected using i2cdetect.

pi@pizero:~ $ i2cdetect -y 1

>    0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
> 00:          -- -- -- -- -- -- -- -- -- -- -- -- --
> 10: -- -- -- -- -- -- -- -- -- -- -- -- 1c -- -- --
> 20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- 6a -- -- -- -- --
70: -- -- -- -- -- -- -- 77

JdaieLin, 2021-03-06:
@wlmh110

jan94, 2021-04-08:
I have the same issue.
Is there anything we can do to counteract that problem ?

jan94, 2021-05-03:
Any progress so far ?

fabiantechera, 2021-06-25:
Hello! I have the same problem with PiSugar 2 Pro and a BME280 sensor. I do not get data from the BME sensor and also PiSugar doesn`t show information about charging and anything else. 
Some help to get both working independently?
Thank you.

Some additional information:

It supposed that BME use the 0x76 address

While Pisugar and BME280 is connected I get this from the console:

pi@raspberrypi:~ $ sudo i2cdump -y 1 0x32
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
10: XX XX XX XX XX XX ^C
pi@raspberrypi:~ $ sudo i2cdump -y 1 0x75
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX ^C
pi@raspberrypi:~ $ sudo i2cdump -y 1 0x75
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
10: XX XX XX XX 

pi@raspberrypi:~ $ sudo i2cdump -y 1 0x76
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX

When BME280 not connected, only PiSugar Connected:

pi@raspberrypi:~ $ sudo i2cdump -y 1 0x32
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 12 01 a1 04 01 04 61 00 00 00 00 00 00 00 00 00    ??????a.........
10: 10 00 00 00 00 00 20 00 00 00 e0 0f 00 00 08 39    ?..... ...??..?9
20: 11 97 06 20 03 61 38 81 04 18 03 61 51 cf 79 fc    ??? ?a8????aQ?y?
30: cd f2 0a b7 9f 71 df 22 b5 6a 02 35 bb 49 7d df    ?????q?"?j?5?I}?
40: bd 7f bd b9 6d 35 5f 17 4f 56 6e 57 40 6f f3 f7    ????m5_?OVnW@o??
50: e6 d6 ff db c3 82 20 ce d5 54 ad 78 9a cf f3 c9    ??.??? ??T?x????
60: 6e cb 7b 76 fd f9 ec df 75 1d e1 9d 1c 7e 3e e6    n?{v????u????~>?
70: 76 3b 20 11 25 32 20 44 10 34 ff ff ff ff ff ff    v; ?%2 D?4......
80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................

pi@raspberrypi:~ $ sudo i2cdump -y 1 0x75
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: ea ff 77 26 bf 00 00 87 03 d0 25 00 fd 1e 04 03    ?.w&?..???%.????
10: 65 df 76 d5 0b 61 03 00 a0 be 00 b8 f8 20 00 00    e?v??a?.??.?? ..
20: ff 8a 92 08 d4 60 a6 63 4a b7 df 8f 28 00 00 00    .????`?cJ???(...
30: 3f 26 1a 70 d8 89 c3 c0 ba dc 11 40 d8 02 ad c0    ?&?p???????@????
40: 19 3d 11 57 55 be d4 e9 00 00 00 00 00 00 00 00    ?=?WU???........
50: 05 56 46 71 02 00 04 00 00 00 00 80 00 00 00 00    ?VFq?.?....?....
60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
70: 37 ba 00 00 08 82 c0 58 00 00 00 26 03 13 08 48    7?..???X...&???H
80: 7f 8c a6 d0 5a 76 b6 87 b0 7f 8c a6 d0 5a 76 b6    ????Zv???????Zv?
90: 87 b0 53 66 86 af 56 86 be 5f 91 53 66 86 af 56    ??Sf??V??_?Sf??V
a0: 86 be 5f 91 b5 3a e9 7e 00 00 00 00 00 00 00 00    ??_??:?~........
b0: f0 93 93 b3 51 7e 00 00 00 00 00 00 00 00 00 00    ????Q~..........
c0: 8d 55 f7 fa 63 23 06 f6 eb fa 00 00 00 00 00 00    ?U??c#????......
d0: ba 0d cb 3e 9a 26 e2 01 00 00 00 00 c2 38 7a 0e    ???>?&??....?8z?
e0: c0 00 00 81 1b 00 00 53 09 00 00 00 11 8d 10 00    ?..??..S?...???.
f0: 08 43 02 61 0a 00 00 70 01 0b 10 b9 00 00 00 00    ?C?a?..p????....

pi@raspberrypi:~ $ sudo i2cdump -y 1 0x76
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
10: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
20: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
30: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
40: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
50: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
60: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
70: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
80: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
90: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
a0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
b0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
c0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
d0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
e0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
f0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX

closed 2024-08-25 by github-actions[bot]