PiSugar/PiSugar #32: Issue with Sense HAT in combination with PiSugar 2 Pro

Gerald-Dicker, 2021-02-18:
When I install the PiSugar 2 Pro in combination with Sense HAT on my Raspberry Pi 4 Model B, none of them work anymore. 

When I run "i2cdetect -y 1", it returns all 128 possible addresses with both devices installed. However, with only Sense HAT installed on the Pi "i2cdetect -y 1" correctly returns: 1c, 46, 5c, 5f, 6a, and with only the PiSugar installed, it correctly returns: 32 and 75.

According to what I found on the internet, Sense HAT uses GPIO pins 2 and 3 for the SDA and SCL signals. Apparently also PiSugar2 Pro uses these two pins for SDA and SCL, but not sure which pin (2 or 3) for which signal...

Does anybody have an idea of the potential problem and solution?

Regards,
Gerald

JdaieLin, 2021-03-06:
@wlmh110

eplamers, 2021-04-07:
I'm running into this issue as well. Interested to find a solution, will start digging through PiSugar & SenseHat code

Gerald-Dicker, 2021-04-07:
Hi, the PiSugar team already know what the problem is and will fix it with their Version 3, expected in H1 this year. Hope this helps

Jdec2004, 2021-04-18:
I'm having the exact same problem with the yahboom cooling hat.

lmf-github, 2021-08-30:
Any  update on if this issue with the SenseHat was solved (and what is referred to in the above post- Version 3 of what, the driver?).
Have an issue with another Hat, and wondering if the solution to this SenseHat issue will fix this too, and if so how to apply the fix?.

RedHandShepard, 2021-11-10:
Yeah i would love an update as well. V3 of what? Is it software? or did i waste 60 bucks just to have to buy another newer version they didn't botch the hardware on?
If its a hardware thing witch one works?

RedHandShepard, 2021-11-10:
Just realized these others asked their questions months ago and didn't even het an answer. That's sad. Guess the pisugar team doesn't care that they sold us garbage advertised as leaving the gpio free for projects.
As far as im concerned yall owe us refunds

wlmh110, 2021-12-02:
> Any update on if this issue with the SenseHat was solved (and what is referred to in the above post- Version 3 of what, the driver?). Have an issue with another Hat, and wondering if the solution to this SenseHat issue will fix this too, and if so how to apply the fix?.

@RedHandShepard 

Sorry, this question is the same as another issue and has not been answered in time
Please contact the after-sales team at the following email address for assistance
pisugar.zero@gmail.com

joemeaux22, 2022-03-19:
Hello. Sorry to dredge up a dead issue. Did the questions above ever get answered? The response above mentioned it is the same as another issue, can someone point me to that issue?

RedHandShepard, 2022-03-25:
Not solved. Waiting on new version of product.

joemeaux22, 2022-03-25:
Thank you for the reply. I'll keep an eye out for next version.

closed 2024-08-24 by github-actions[bot]