PiSugar/PiSugar #34: PiSugar 2 Pro will not turn on

stphnduvall, 2021-03-02:
When I got the PS2Pro a few weeks ago I had an issue when turning the board on. No LED's would turn on and it would not give power to the raspberry pi. I broke out my multimeter and check that the battery had enough voltage and it did. After trying many different things it all of a sudden it just started working. I don't know what I did

Today I unplugged the battery and after plugging it back in I am having the same issue. I have it plugged in and charging right now but before I unplugged it it was around 60%, I feel like that should not be the problem. 

Any suggestions or help would be appreciated. Thanks.

stphnduvall, 2021-03-02:
Found that there is an "Activate" button which I had believed I already pressed. I'm not really sure why there is an on/off switch and a button as well. Its not very well documented either. In the readme its written under the "Features" section as "On-board Switch" but on the board its labeled "Activate."

closed 2021-03-02 by stphnduvall

Tetsujinfr, 2022-03-14:
Oh boy, @stphnduvall you are my hero of the day. I was running nuts not finding the issue with my pisugar. Then thanks to your post I rediscovered that activate button, and immediately the i2c 75 adress of the pisugar appeared again and it worked back fine.

In case it helps anyone else, in my scenario I left my rpi and pisugar turned on during the night, and not sure why but next morning all was off.
Not sure what happen, maybe my regular charger was switch off and my pisugar got drained.
Anyhow, it would not turn back on using the power on button no matter what, until I press the activate button.
So if your system is in a case all wrapped up carefully in a case and you have this issue, I sympathise.