PiSugar/PiSugar #37: Custom Shell not executing on button tap (any kind)

benderprod, 2021-03-20:
I've set the single tap button to Custom Shell.

I entered the following:  ~/bender/start_usb.sh

I also tried: sh ~/bender/start_usb.sh

I also tried making the script executable.

The script does not execute.

JdaieLin, 2021-04-10:
Hi, please check the follow:
1. choose the right model when you install PiSugar-server. If you purchased PiSugar2 (not Pro) in 2021, it should be 2-led version.
2. open the management webpage to check whether the server can receive button events. If can, there should be a green tag show up when you press the button.
3. make sure you can run the shell by normal user. Try add sudo before the script.

sjdevlin, 2021-06-08:
I think you need to provide the whole path rather than using the tilda shortcut (e.g. /home/pi/bender/start_usb.sh).  The reason I think is that the ~ reference is not the same for the process running the pisugar server and from which the shell script is launched.  I made the same mistake.   Once I replaced the tilda sign with /home/pi - it worked fine.

closed 2024-08-22 by github-actions[bot]