PiSugar/PiSugar #39: Battery swaps and temperature monitoring

mjbedford2017, 2021-04-07:
Hello, more of questions than an issue. First of all, love the product. When I bought the PiSugar 2 Pro, it mentions that because the connector is a standard PH2.0, battery swaps are possible. This product comes with a 5000mAH battery pack as you know. However, there are 10000mAH packs out there. Could I simply buy one and plug it in instead and effectively double the run time?

Also, can you wire two battery packs in parallel to double the capacity? 

Finally, is there any kind of support to temperature monitor the battery for safety during charging? I don't see a temp probe but can one be added?

Thanks!

JdaieLin, 2021-04-10:
Hi, sure you can use a 3.7v 10000mAh lithium battery to connect to PiSugar 2 Pro thought PH2.0. But using two battery packs in parallel is not suggested. Since the management chip can not recognize there are multi batteries and it will keep charging if any one of the battery is not full, which is dangerous for the full one. For the final question, PiSugar had no support for temperature monitor, if the chip reach over 100 C it will shut down.

4bitFox, 2021-06-10:
In case if there's ever going to be a new revision, I think temperature monitoring could be taken into consideration as it would improve safety and the lifespan of the battery.

JdaieLin, 2021-06-20:
We will use an MCU with temperature monitoring in the next version. But it just monitors the chip itself. Monitoring battery temperature requires lithium battery with NTC feature, which it will cost more and also not easy to find on market. So we may not consider this.

closed 2024-08-16 by github-actions[bot]