PiSugar/PiSugar #41: Battery level fluctuates

4bitFox, 2021-04-25:
The battery level fluctuates when discharging.
I checked here every few seconds:
```
pi@FengCam:~ $ echo "get battery" | nc -q 0 127.0.0.1 8423
battery: 49.48841
pi@FengCam:~ $ echo "get battery" | nc -q 0 127.0.0.1 8423
battery: 50.2772
pi@FengCam:~ $ echo "get battery" | nc -q 0 127.0.0.1 8423
battery: 49.16612
pi@FengCam:~ $ echo "get battery" | nc -q 0 127.0.0.1 8423
battery: 47.204803
pi@FengCam:~ $ echo "get battery" | nc -q 0 127.0.0.1 8423
battery: 46.330696
pi@FengCam:~ $ echo "get battery" | nc -q 0 127.0.0.1 8423
battery: 46.019882
pi@FengCam:~ $ echo "get battery" | nc -q 0 127.0.0.1 8423
battery: 45.803726
pi@FengCam:~ $ echo "get battery" | nc -q 0 127.0.0.1 8423
battery: 45.441277
pi@FengCam:~ $ echo "get battery" | nc -q 0 127.0.0.1 8423
battery: 45.062836
pi@FengCam:~ $ echo "get battery" | nc -q 0 127.0.0.1 8423
battery: 44.98405
pi@FengCam:~ $ echo "get battery" | nc -q 0 127.0.0.1 8423
battery: 45.1733
pi@FengCam:~ $ echo "get battery" | nc -q 0 127.0.0.1 8423
battery: 44.594387
pi@FengCam:~ $ echo "get battery" | nc -q 0 127.0.0.1 8423
battery: 44.396088
pi@FengCam:~ $ echo "get battery" | nc -q 0 127.0.0.1 8423
battery: 45.29474
pi@FengCam:~ $ echo "get battery" | nc -q 0 127.0.0.1 8423
battery: 46.06507
pi@FengCam:~ $ echo "get battery" | nc -q 0 127.0.0.1 8423
battery: 48.155525
pi@FengCam:~ $ echo "get battery" | nc -q 0 127.0.0.1 8423
battery: 46.497635
pi@FengCam:~ $ echo "get battery" | nc -q 0 127.0.0.1 8423
battery: 46.322018
pi@FengCam:~ $ echo "get battery" | nc -q 0 127.0.0.1 8423
battery: 47.18013
pi@FengCam:~ $ echo "get battery" | nc -q 0 127.0.0.1 8423
battery: 48.177963
pi@FengCam:~ $ echo "get battery" | nc -q 0 127.0.0.1 8423
battery: 48.317566
pi@FengCam:~ $ echo "get battery" | nc -q 0 127.0.0.1 8423
battery: 48.128498
pi@FengCam:~ $ echo "get battery" | nc -q 0 127.0.0.1 8423
battery: 48.45958
pi@FengCam:~ $ echo "get battery" | nc -q 0 127.0.0.1 8423
battery: 48.00657
```
When I plug it in to charge the level shoots from 45% - 50% up to 85% and the level is more stable (around +/-0,5%)
Connected to the Pi 4B is the HQ Camera Module and HDMI. I'll hook up a screen later.

JdaieLin, 2021-06-08:
It's a normal behavior. Since the battery level is a curve base on the output voltage, which will suddenly goes up when you plugin and start charging with big current, and leads to the fluctuates.

4bitFox, 2021-06-10:
I know, it's details but it's just weird to see the battery charge go up and down even when not plugged in. 
That's because I display the battery level in my UI that I made for shooting Pictures.

Maybe a nice "enhancement feature" would be, if the software displays a value that is averaged out over a given time like 3 minutes or something?

I don't know if anything regarding the battey level shooting up can be done, but it would be nice because if it would show the actual charge. 
Imagine plugging in the Phone, see it charging up to 80%, unplugging it and seeing later that the battery has "discharged" by 40% in no time.

JdaieLin, 2021-06-20:
Honestly I don't know how normal phones deal with this problem.
We will try to look into this and make a better curve. Maybe calculating an averaged number, considering the charging status is a way of enhancement on software.

Tetsujinfr, 2022-03-14:
Yeah I observed the same volatility. Adding a simple ewma filter would likely make it appear more "robust". It only requires the new current charge value and the previous estimate value to compute a new estimate. You can select how much information the new value contributes to the new estimate through an alpha parameter, e.g. 0.5 for a 50/50 contribution between history estimates and the new value. Simple and more effective/flexible than simple averaging. Just my two pennies.
Great product/idea anyway. Continue the good work guys!

Docfch, 2024-03-29:
I tested with pisugar 3 plus version 1.7.6, i have the same issue with @4bitFox.
The battery level fluctuates when charging and discharging.
And also, when I plug it in to charge the level shoots from 30% up to 50%

closed 2024-08-14 by github-actions[bot]