PiSugar/PiSugar #42: Fragile clips on caps

hijackeel, 2021-04-26:
My 3D-printed PiSugar2 case from Treatstock just arrived, with the caps already snapped in place to the shell:

1) https://github.com/PiSugar/PiSugar/blob/master/model/pisugar_case_common_cap.STL
2) https://github.com/PiSugar/pisugar-case-pihat-cap/blob/master/2.13inch_epaper_cap/pisugar_case_epaper_cap.STL
3) https://github.com/PiSugar/PiSugar/blob/master/model2/pisugar2_shell_xl.STL

While removing the common cap from the shell, one of the clips snapped off. :(
I could order a replacement, but I'm afraid it would just break again.
Could the caps be redesigned with more robust clips?

Tetsujinfr, 2022-03-14:
Agreed the caps are very fragile. I had to be very careful when removing supports after the print and some of the caps went off anyway when trying to close the case. The design is good but targeting plastic injection type of process/materials I think, not really 3d printing. It should be strengthened or replaced with a screw/bolts solution.

hijackeel, 2022-05-21:
@Tetsujinfr @JdaieLin  Yes, screws would be a great idea!

closed 2024-08-13 by github-actions[bot]