PiSugar/PiSugar #46: Softbutton Startup

tbaumann, 2021-06-15:
I am a little surprised that I can apparently shutdown the device via the programmable button but not wake it up.
That's honestly what I thought I could do when I read the term "programmable button".

Right now I can nicely shutdown the pi, but I still have to slide the clunky power button off and on again to boot back up.

Am I missing a hint or is the button actually not able to wake the power management chip?

JdaieLin, 2021-06-20:
Hi tbaumann, yes the button cannot wake it up since its programmable feature relays on the pi. On PiSugar 2 we can only wake it up by the RTC. In the coming S and 3 version we will make the button capable to wake up itself.

JdaieLin, 2021-06-20:
You can also wake up the chip by manually turn off and on the power switch.

bongoplayingmonkey, 2021-06-27:
Is there anyway to solder in a momentary switch to temporarily ground that switch and simulate cycling the power button? Feels like that should be the solution but it's difficult to work out without the schematic.

JdaieLin, 2021-07-18:
@wlmh110

closed 2024-08-09 by github-actions[bot]