PiSugar/PiSugar #47: External switches.

bongoplayingmonkey, 2021-06-25:
I need to install the pisugar 2 pro into a case where the power switch isn't accessible. Any tips for adding in an external switch to solve that issue?

bongoplayingmonkey, 2021-06-27:
Can there at least be some JST sockets to extend the switches out to surface mounts on future versions?

JdaieLin, 2021-07-18:
@wlmh110

GSVNoFixedabode, 2021-08-11:
Any update on this?  I have the same requirement.

wlmh110, 2021-08-16:
Here is the PCB of the PiSugar 2 Pro power switch.
[image]
When the 2(VOUTC1) pin is connected with 3(VOUT1), PiSugar will work, and when the 2(VOUTC1) pin is connected with 1(GND), it will shut down.

fheslouin, 2021-08-31:
@wlmh110 where can we find those pins ? PCB is not showing any of these names

sledgehammer617, 2022-03-01:
Any updates on this? Anyone figure out how to do it? 

I was also hoping to install my pisugar 2 pro in a case with an external button.

Acluff1, 2022-03-15:
I figured it out! Simply solder your toggle switch wires to "Ctrl" and "Off," leave the main switch in the "On" position and bam! 
[image]

sledgehammer617, 2022-03-16:
@Acluff1 Cool, thanks!

sjin4903, 2022-07-19:
@HaochengZheng 
I want to know exactly where you're connecting.

HaochengZheng, 2022-07-19:
After you remove the metal cap over the switch and the plastic handle, there are 3 metal areas underneath. Originally, when the switch is at "off", middle and left metal areas are connected; when the switch is at "on", middle and right part is connected. So get a slide switch, solder 3 pins of slide switch to each metal area(right pin- right, middle-middle, left-left)

HaochengZheng, 2022-07-19:
@sjin4903 
After you remove the metal cap over the switch and the plastic handle, there are 3 metal areas underneath. Originally, when the switch is at "off", middle and left metal areas are connected; when the switch is at "on", middle and right part is connected. So get a slide switch, solder 3 pins of slide switch to each metal area(right pin- right, middle-middle, left-left)

sjin4903, 2022-07-19:
@HaochengZheng 
Problem solved! Thanks

stokes776, 2022-07-29:
Any idea if I want to just connect a switch that has two leads, not three?

OFF = Middle and LEFT so...... X X o
ON = Middle and Right          o X X

Feels like maybe I could just ignore that orange wire you have in your photo?

HaochengZheng, 2022-07-29:
@stokes776 I don't think so. I guess there is some kind of capacitor, so you have to make MIDDLE and LEFT pad connected before switching to MIDDLE and RIGHT

spaceman1988, 2023-04-29:
> I figured it out! Simply solder your toggle switch wires to "Ctrl" and "Off," leave the main switch in the "On" position and bam! I should add this was a pisugar S plus.
> 
> [image]

I have an "S". The document isn't very clear on what the CTRL-OFF-ON pinouts actually do. Do the pinouts control power going to the Pi (turning power on and off), or do the pins set the state of the Pi's GPIO3?

I have a Pi Screen installed and can safely shutdown the Pi from the screen, but since it's in a case I need a way to turn in back on. Referring to your photo, I'm assuming these pins are controlling power to the Pi your toggle switch in one direction would take power away from the Pi and flipping it the other way would turn it back on. Correct? If all I wanted to do was turn it back on from screen-triggered shutdown then a momentary pushbutton between CTRL and OFF should do the trick.

closed 2024-08-08 by github-actions[bot]