PiSugar/PiSugar #48: what's the tolerable tempreture of PiSugar 2 Pro 5000mAH battery?

TyphoonWu, 2021-06-27:
Question as the title, some time the raspberrypi 3b+ work tempreture will increase into 70°C~80°C, is that the PiSugar 2 Pro 5000mAH battery able to work well and not bomb!?

JdaieLin, 2021-07-18:
@wlmh110

fheslouin, 2022-03-09:
I'm using Rasp 4 and Pi Sugar 3 Plus. The device is going to run under the tropic, an average of at least 30°C all year round. 
Optimum condition for the battery would be 20°C 25°C working temperature and not above 45°C during charging. I've seen that the test certificate shows that the battery has been exposed to 72°C during 6 hours. 

The current design of the Pi Sugar suggest that the battery is "hold" on the back of the Pi Sugar which get really hot especially during battery charge. I measured almost 90°C on the Pi Sugar chip during charging while being on a 35°C ambiance. 

So my advice (what I should have done) is, detach the battery from the rear of the Pi Sugar and put it somewhere else on your case where it can cool down a little bit. Further more, raspberry cooling is better held vertically

Tetsujinfr, 2022-04-03:
@fheslouin totally agree with you.
My experience with a pisugar3 Plus: I ran my rpi to 70C - 75C for some time, in a very large but quite closed case, with no cooling (no fan, very small sink on rpi cpu). The pisugar was plugged to charge if need be.
Next day morning, when I open thebcase for another reason, I found the battery hanging by the cable with no magnet on it anymore: the glue had loosen up due to the heat which somehow spread to the magnets and then the glue.

Now I am fixing the battery not under the oisugar pcb, and I have active cooling in place.

@TyphoonWu  Just to say that the magnet thing below the pisugar pcb is a cool diy idea under nice temperatures, but not good for long term, always running in enclosed cases, type of scenarios.

closed 2024-08-08 by github-actions[bot]