PiSugar/PiSugar #49: Procedure for shutting down and starting up

OctaneInteractive, 2021-07-04:
Hi!

I have a PiSugar 2 Pro, and I'm a bit confused about the correct procedure for shutting down and starting up again.

The first time I used it, I first pressed the "Activate" button for 2 seconds, then flicked the switch.

When shutting it down, I first shut down the OS, then flick the switch.

When starting up again, do I have to press the "Activate" button for 2 seconds before flicking the switch?

I think there should be a table explaining the different button and switch combinations and under what conditions.

JdaieLin, 2021-07-18:
Hi, normally there is not need to press active button unless the battery was run out. Activation is some kind of protective schema of the Injoinic chip, and it's also not well documented in the chip datasheet. So we decide to use a replace the chip, avoid Injoinic. If you still need need to activate every time before boot, please contact pisugar.zero@gmail.com for replacement. Thanks!

psi5asp, 2021-08-12:
No need to touch the PiSugar when shutting down the RPi (unless as stated above, if the battery runs dry). 

The PiSugar will keep its lights on when the RPi lights are off (RPi = shutdown).

Kung-Foo-Kamel, 2021-08-23:
To power you Pi back on after it has been Halted or safely powered down, just momentary short Pin 5 to Pin 6 on the GPIO header, this will power the Pi back on.

You can make a small button and fix it to you case for easy access.

fheslouin, 2021-08-31:
Shorting Pin 5 & 6, has no effect. Did you manage to have it working ?

Kung-Foo-Kamel, 2021-09-02:
Yes it works perfectly for me, I send the halt or poweroff command through SSH, and I have a momentary press button connected to pins 5 and 6, one press and the Pi fires back up.

The Pi must already be up and running via the initial start-up procedure for the PISUGER2 Pro, but after that it will power up perfectly for me.

Below is an image of my custom Pi Cluster and one operational module with the start-up button installed.

[image]

fheslouin, 2021-09-13:
Yes, you are right, I had probably mess something ! Thanks

funkyfourier, 2022-05-03:
I have not been able to power up the Pi shorting pins 5 and 6 when connected to the Pi Sugar 2 Pro. When powered from the wall socket USB plug I notice that after shutting down the red power indicator light on the Pi stays lit, and in that situation shorting pins 5 and 6 powers it on. However, when powered from the Pi Sugar, the red power indicator light turns completely off after (safely) shutting down the Pi. 

Should there not be some power coming from the Pi Sugar at all times? Or is it supposed to cut completely off like this?

NexGen-3D-Printing, 2022-05-03:
Mine does the same, if you pull power from the Pi Sugar and shutdown the RPI, you will need to power everything backup from the button on the Pi Sugar.

Seems like an obvious design flaw, they should have just provided some pins for a simple momentary button input that could both power up and power down the RPI through the Pi Sugar, regardless of the external power input status (provided there is sufficient battery charge).

This is an example of another great product idea, that is "almost" there, they got to about 80% and launched it, good idea, good product, but missing a couple of important key features still.

On a side note, keep your eye on the battery pack, I purchased 6 of these, setup 5 of them, 3 out of the five resulted in swollen battery packs that started to warp the PLA enclosures I made, and I had to discard them as they were on the verge of bursting/combusting, extremely bad for a device that could be locked away as an IOT like product, I will be swapping all the packs out for some good quality Samsung/Panasonic 18650's, not a fan of these pouch cells.

closed 2024-08-06 by github-actions[bot]