PiSugar/PiSugar #50: Issue with other I2C devices

tamugo, 2021-07-22:
I am working with two Qwiic sensors connected to a raspberry pi zero via a qwiic pHAT. When the pHAT is not connected the I2C addresses of the PiSugar2 can be detected and data can be read from it. However, when I connect the pHAT, the addresses of the PiSugar are no longer detected but the addresses of the two sensors are. Any help in solving this would be really appreciated.

JdaieLin, 2021-07-29:
@wlmh110

ahmed-mans, 2021-08-08:
I had a similar problem where I had an infrared camera working on I2C and my PiSugar2 Pro with a Rpi 4B. I could not use both devices when they were connected to the Rpi. Normally they should have cause their I2C adresses were different but somehow there was a problem. What I did to solve it was to just choose another I2C bus for my infrared camera and changed the software accordingly and it worked.

Hope it can help you ;)

chickenherb, 2021-09-09:
I have similar problems with my PiSugar2 and I2C comunications.

When I use my raspberry pi zero w with PiSugar2 only it works fine:
pi@zeropi:~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- 32 -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- 75 -- --

When I use my Waveshare 4-Channel INA219 HAT without PiSugar on my raspery pi zero w, it works fine too:
pi@zeropi:~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: 40 41 42 43 -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

But when I use both the PiSugar together with Waveshare 4-Channel INA219 HAT on my raspberry pi zero w,
the I2C bus doesnt work any more. So the Waveshare 4-Channel INA219 HAT and the PiSugar2 on my raspberry pi zero w
is not working anymore:
pi@zeropi:~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- 04 -- 06 07 -- 09 0a -- 0c -- 0e 0f
10: -- -- 12 13 14 -- -- 17 18 -- -- -- 1c 1d 1e --
20: -- 21 22 23 -- -- 26 -- -- -- 2a 2b 2c -- -- 2f
30: 30 31 -- 33 34 -- -- -- 38 -- -- -- 3c 3d 3e --
40: 40 41 42 43 -- 45 46 47 -- -- -- 4b 4c -- -- 4f
50: 50 -- -- -- 54 55 56 -- -- 59 5a -- -- -- 5e 5f
60: -- -- 62 63 64 -- -- -- 68 69 -- -- -- -- 6e 6f
70: 70 -- 72 73 74 -- -- --
Is there any solution for my problem ?
Thanks in advance !

ahmed-mans, 2021-09-10:
Normally, as the i2c adresses are different it should be working fine. In my case I put the devices in different i2c ports and it worked. Now I don't know if it is possible in your case since you have a HAT device.

chickenherb, 2021-09-10:
Thank you for your reply,
as you can see the I2C device adresses are different. My hat uses 40, 41,
42 and 43.
This is different to the PiSugar2 used ones.
But when I scan the I2C bus with bot devices mounted then the bus scan
results in crazy found adresses and I can Not operate any I2C device.

Tanks a lot for your reply

Chickenherb

jonhaun, 2022-01-11:
I've experienced this too. (PiSugar2, RPI zero W, [https://ozzmaker.com/berrygps-berrygps-imu-quick-start-guide/](BerryGPS-IMU))

The IMU uses:
- 0x6A for the gyroscope and accelerometer
- 0x1C for the magnetometer
- 0x77 for the pressure sensor
- 0x42 for GPS

The RPI+Sugar works great on its own. As does the RPI+IMU. 

Combining them (RPI+Sugar+IMU) causes weird behavior. If I call i2cdetect multiple times. The IMU registers seem to persist, but disappear intermittently, and register 75 for the Sugar is pretty much always gone, 

I'd appreciate any help. My next debug focus will be the baud rate.

Tetsujinfr, 2022-04-03:
> Normally, as the i2c adresses are different it should be working fine. In my case I put the devices in different i2c ports and it worked. Now I don't know if it is possible in your case since you have a HAT device.

@ahmed-mans what do you mean by "I put the devices in different i2c ports and it worked" ? On different i2c pins? If yes which ones please?

Thanks

closed 2024-08-06 by github-actions[bot]