PiSugar/PiSugar #51: Pi suçât pro 2

coutud, 2021-08-09:
Hi i have bought a pi sugar for my pi 4 i am vert glad but i don’t know when she is charge because With the Power my battery is always at 26%
the Light Blue is switch on
But i don’t know why sole Time i have Green light

psi5asp, 2021-08-12:
If you have installed the PiSugar deb (see wiki install instructions), then try using 'PiSugar 2 Pro' (`dpkg-reconfigure pisugar-server`) to view the charge % in the webpage.

GSVNoFixedabode, 2021-11-14:
The 26% figure (actually 26.378838) seems to indicate a Compatibility issue: I get the same reading when using a PiSugar2 Pro board with a Pi Zero: setting it to the Pro, and the board shuts down immediately (battery 0%), but setting it the PiSugar2, it shows as 26% but remains static no matter board charge level.  Note board itself charges battery ok (goes from blue LED to green when fully charged).  I have yet to find a version' setting that allows the Zero to use a PiSugar2 Pro board. This on 1.4.9, 1.5, and 1.6.1 versions of software running on Pi Buster.

xlla, 2022-03-04:
PiSugar2 Pro with Pi 0 on 1.6.4 also has this problem, the board shuts down less than 1 minutes once power on.

closed 2022-04-28 by coutud