PiSugar/PiSugar #52: PiSugar has slow reponse for command queires

w1v1, 2021-08-16:
I have a few Pi4's with DietPi buster, and all work fine with PiSugar.

I got a new Pi4 and installed DietPi bullesye and the PiSugar is very sluggish to respond to commands. The settings are duplicated with the workine ones, so is there some issue with bullseye (that's the only difference)?

I swapped the PiSguars from the Bullseye Pi4 with the Buster Pi4 and there was no issue with the h/w, its just when PiSugar is used with Bullseye it appears to be the problem.

+-------- Pi Buster Settings --------+
                 Serial/UART      : Manage available devices                       │
│                 Bluetooth        : [Off]                                          │
│                 I2C state        : [On]                                           │
│                 I2C frequency    : [100 kHz]                                      │
│                 SPI state        : [Off]                                          │

+-------- Pi Bullseye Settings --------+
     Update RPi4 EEPROM firmware : [Bootloader: 1619712685 | VL805: 000138a1]      │
│     Serial/UART                 : Manage available devices                        │
│     Bluetooth                   : [Off]                                           │
│     I2C state                   : [On]                                            │
│     I2C frequency               : [100 kHz]                                       │
│     SPI state                   : [Off]                                           │

Any other settings I should check on PI4 Dietpi Bullseye?

w1v1, 2021-08-17:
Same issue on Pi-4 and Pi-3, both bullseye.

JdaieLin, 2021-09-20:
@fengyc

fengyc, 2021-09-22:
Working on it ~

closed 2024-08-06 by github-actions[bot]