PiSugar/PiSugar #53: When triggering custom button function: getting error

snarflakes, 2021-08-20:
Looks like a possible conflict between homebridge server and pisugar server when triggering the pi-sugar button.  Has anyone seen this before? This is what comes up on journalctl read out after pressing the pisugar accessory button.

Aug 19 17:41:01 homebridge pisugar-server[305]: [2021-08-20T00:41:01Z INFO  pisugar_server] Websocket closed

thanks,
Snarflakes

wlmh110, 2021-09-23:
Hi,
Can you take out "sudo" and try again?
just:
python3 /home/pi/ganpunks/NFTGIFDisplay/QRCODElinker.py

closed 2024-08-06 by github-actions[bot]

Zanop, 2024-11-13:
Hi,

I am getting the same issue, custom button commands generally work but I get these lines in the log constantly:

Nov 13 13:30:27 Curigotchi pisugar-server[243]: [2024-11-13T12:30:27Z ERROR pisugar_server] Invalid
Nov 13 13:30:41 Curigotchi pisugar-server[243]: [2024-11-13T12:30:41Z ERROR pisugar_server] Invalid
Nov 13 13:30:53 Curigotchi pisugar-server[243]: [2024-11-13T12:30:53Z ERROR pisugar_server] Invalid
Nov 13 13:31:04 Curigotchi pisugar-server[243]: [2024-11-13T12:31:04Z ERROR pisugar_server] Invalid
Nov 13 13:31:15 Curigotchi pisugar-server[243]: [2024-11-13T12:31:15Z ERROR pisugar_server] Invalid
(END)