PiSugar/PiSugar #58: Pi Sugar 2 Pro Powers Off After 1 Minute

deanmarano, 2021-10-16:
I have a Pi Sugar 2 Pro powering a Raspberry Pi Zero - I realize the wrong fit, but I wanted the larger battery and RTC.

WHen I power the PiSugar from micro USB, it powers off after less than a minute. The system does not shutdown, it just turns off.

After I press the "activate" button, it turns on, and then turns off after less than a minute. Same as when initially plugged in.

It never stays on for more than a minute. I have had the same results when powering over USB-C.

Any idea why this is happening?

JdaieLin, 2021-10-17:
Looks like a issue of pisugar-poweroff ?
@fengyc

GSVNoFixedabode, 2021-11-17:
Same pairing here (Pro + Zero) and same issue - mgmt software is powering the Pi off immediately as battery detection is 0%.  If you re-configure the software to set the model to 'PiSugar2', can you confirm that it works, but displays '26%' as the battery level?

fengyc, 2021-11-18:
> Same pairing here (Pro + Zero) and same issue - mgmt software is powering the Pi off immediately as battery detection is 0%. If you re-configure the software (_dpkg-reconfigure pisugar-server_) to set the model to 'PiSugar2', can you confirm that it works, but displays '26%' as the battery level?

@GSVNoFixedabode Edit /etc/pisugar-server/config.json, set `auto_shutdown_level` to 0.0 and `auto_shutdown_delay` to a  big number like 9999999.0 . 

Could you show us the  `i2c dump -y 1 0x75` result, that would help us to figure out what went wrong.

GSVNoFixedabode, 2021-11-18:
[image]

jcothran, 2022-01-10:
@fengyc Having the same issue as @GSVNoFixedabode , editing config.json values did not seem to change the unit from auto shutdown after about a minute.

fengyc, 2022-01-11:
@jcothran I am trying to fix this.  You could download and install the latest development build, set `auto_shutdown_level` to null.

jcothran, 2022-01-11:
thanks for checking, I tried the latest development 1.6.5 after changing some of the hardcoded cdn/channel references which failed to 
https://github.com/PiSugar/pisugar-power-manager-rs/releases/download/latest/ Unfortunately it shows the battery at 100% and powers off the pi shortly after. Checked the config.json file and auto shutting down while auto_shutdown_level was set to null. Can keep it on with the previous version and auto_shutdown_level set to null, but does not read battery, saying 'invalid request' with 'get battery'

I'm using the 'Pisugar2 Pro' install setting, not sure if I should try '2-LED' instead? The version I ordered last month does have 2 LED's, one for the charging status and one for the powering status, but the ordering and packaging don't mention this product suffix so not exactly sure of the version I have. thanks

xlla, 2022-03-03:
same issue here, and
<pre>
pi@rpi0:~ $ sudo i2cdump -y 1 0x75
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
10: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
20: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
30: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
40: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
50: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
60: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
70: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
80: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
90: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
a0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
b0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
c0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
d0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
e0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
f0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
</pre>

GSVNoFixedabode, 2022-07-08:
Update: Pi Sugar2 Pro mated with Pi Zero.  Just installed version 1.7.1 (uninstalled old version first), specifying 'Pi Sugar 2 Pro' as the model.  Battery now reports correct level!  Great to see this bug fixed.

closed 2022-07-18 by JdaieLin