PiSugar/PiSugar #59: PiSugar 2 & On Power Restore

jpcetre, 2021-10-20:
Hi,
I am playing with my new PiSugar2 (2 leds) and a Pi Zero W. 
I expected to use it as an UPS and found the On Power Restore option for the Schedule Wake Up, but I am not able to make it work. 
Any trick I should know?
(I tried to plug sugar and pi on 2 separate chargers but same result)

[image]

JdaieLin, 2021-11-28:
@wlmh110

wlmh110, 2021-12-03:
Hi,
Can it start up once? Or does it never starts up after shutdown?

erlendv, 2022-01-14:
@wlmh110 Im experiencing the same thing. Just purchased pisugar2. I selected the on power restore option. After shutting the pi down, disconnected and reconnected power, nothing happens. A couple of things I noticed:
- after booting the pi (turning pisugar off and on) and going into the web interface a warning pops up that I have turned off wake-on-power-restore and the setting in the UI turns back to disabled
- however, in /etc/pisugar-server/config.json `auto_power_on` is `true`

xlla, 2022-03-04:
I don't know where to ask a question, so post it here:
Can I use same power supply to charge the Pi and PiSugar while using feature "On Power Restore" ?
My purpose is when I turn on the power my device and Pi & PiSugar will be power on and start charge the PiSugar, when I turn  off the power , Pi will notice that and power off auto.

wlmh110, 2022-03-05:
> I don't know where to ask a question, so post it here: Can I use same power supply to charge the Pi and PiSugar while using feature "On Power Restore" ? My purpose is when I turn on the power my [...]

You need to use different chargers, otherwise PiSugar will charge itself by battery

wlmh110, 2022-03-05:
> @wlmh110 Im experiencing the same thing. Just purchased pisugar2. I selected the on power restore option. After shutting the pi down, disconnected and reconnected power, nothing happens. A couple of [...]

We suspect it is a software bug. What version of software are you using now?

Tetsujinfr, 2022-04-03:
@erlendv do you still experience that issue with the on pwr wake up function? If not, how did you fix it? (Software update, other?)

bluefireexplosion, 2022-06-24:
Still experiencing this issue with a pisugar 2, auto_power_on is enabled. When shutting down the pi, removing power, then reconnecting power, the pi will not turn back on until the power switch is toggled. Any solutions?

wlmh110, 2022-07-05:
> Still experiencing this issue with a pisugar 2, auto_power_on is enabled. When shutting down the pi, removing power, then reconnecting power, the pi will not turn back on until the power switch is toggled. Any solutions?

First, please update the latest software
Secondly, can you describe what happened after the external power supply was restored?

kcmd, 2023-06-10:
> First, please update the latest software Secondly, can you describe what happened after the external power supply was restored?

I'm experiencing the exact same problem described by bluefireexplosion. The latest software is installed. Being able to automatically turn the Raspberry Pi on again after power has been restored is an important capability for remotely installed devices that are being recharged by a solar panel.

kcmd, 2023-06-10:
Likely relates to Issue #98.

closed 2024-07-26 by github-actions[bot]