PiSugar/PiSugar #62: Qi used as 5V VIN but not charging

pauloesperon, 2021-11-10:
Hi there, I would like to use the Qi pads to wire a USB-C female breakout board 5v input to I can put the charging port somewhere else on my enclosure instead of using the micro USB port. I wired it but it is not charging. Any idea what might be happening? 

As far as my wiring, from the break out board I connect the VBUS and GND to + and - Qi. 

Thank you!

JdaieLin, 2021-11-28:
@wlmh110

wlmh110, 2021-12-03:
Could you please take a picture?

dbrb2, 2021-12-31:
Should this work? I was hoping to do the same, but haven't yet wired it up...

wlmh110, 2022-01-04:
As long as 5V power supply is input, it can work

nachotisnes, 2022-02-06:
Did you manage to get this working? I'm facing the same issue. Sorry in advance if I make newbie errors, i'm kinda outside my comfort zone here.

I've been doing some measuring with my multimeter, and I'm getting the idea that the correct wiring may be the opposite one.

My measurements are:
- When the QI adapter is not plugged to the board and is "active", the voltage is 4.92
- When the QI adapter is wired as you mention, the output voltage from the adapter drops to ~0.3, and the adapter starts working erratically: it seems to be turning on and off.
- In that case, the battery is shown as "not charging" and continues discharging using the webUI, although using the command line sometimes I get a battery_charging=true.

@wlmh110 can you make something of this?

wlmh110, 2022-02-07:
Hi, can you take a picture of the system?

nachotisnes, 2022-02-07:
It was a dumb mistake in my part.

Seeing the pics I posted eariler y realized my mistake, I had the wires inverted (VIN/GND).

After correcting that the web UI shows a "Charging" status. Will continue my testing.

Thanks!

closed 2024-07-26 by github-actions[bot]