PiSugar/PiSugar #63: I2C Conflicts with other devices

mlaity, 2021-11-11:
This has been brought up a couple times in the past, but no response from PiSugar support, I've only seen responses from other users in the community.

I'm using a Raspberry Pi 4B, OzzMaker BerryGPS-IMU-4, and PiSugar 2 Pro.  When both devices are connected, the I2C addresses continually come and go for each device, visible via continual i2cdetect scans, making both devices inaccessible. There does not seem to be a conflict in I2C addresses between these devices, with the BerryGPS-IMU-v4 using 1c, 6a, and 77, and the PiSugar using 32 and 75.  This doesn't seem to be an issue with the BerryGPS-IMU. Their support forums only had one complaint of this issue, but it was also in conjunction with a PiSugar.  We have used other I2C devices with the BerryGPS-IMU without trouble.  Seems to be something specific to the PiSugar.

We can't just create additional I2C buses as both devices have built-in headers to connect to I2C Bus 1 on Pi Pins 3 and 5.  This has been the common path forward for others experiencing this problem, but I am just up against the hardware/pin layout of each device.

JdaieLin, 2021-11-28:
Hi, we think it's a limitation of PiSugar2's I2C feature which provided by the power chip. You can contact pisugar.zero@gmail.com for the coming PiSugar3 with an independent MCU.

eqikkwkp25-cyber, 2022-09-10:
Hello, is this kind of issue solved with the available PiSugar3? 

Unfortunately as a PiSugar2 owner i am not very happy with this kind of 'design issues', in my case its an EPaper HAT in combination with PiSugar2 and a Raspberry Pi Zero2: [[Errno 121] Remote I/O error](https://github.com/waveshare/Touch_e-Paper_HAT/issues/10#issuecomment-1141375782). I already checked with another Pi Zero2 but i observe the same issue there. 

Any workaround how i could resolve this?

spaceman1988, 2023-04-29:
If you're adding a HAT that uses i2c (like a sound card) the PiSugar S has a switch that turns off i2c (meaning it doesn't hold CPIO3 (pin 5) in "low" killing i2c communications. That feature works, BUT... if your setup is going in a case accessing the tiny buttons on the PiSugar board to turn the Pi back on isn't possible. You can turn off the Pi by assigned an unused CPIO to a pushbutton to turn off the Pi, but in a case there's no way to turn it back on. Pulling the power doesn't work since battery power is still available. Since most folks put their Pi in some sort of case this is a major issue with the PiSugar's design.

closed 2024-07-26 by github-actions[bot]