PiSugar/PiSugar #65: Invalid slave address: 50 and hwclock: Cannot access the Hardware Clock via any known method.

matthiasg, 2021-12-28:
Not sure these might be related or due to bad soldering of the headers, but are these two issues connected ?

This same battery worked fine on a Zero 1st Gen WH with no issues accessing the clock, but I didn't look into the logs about slave address: 50.

JdaieLin, 2021-12-29:
@fengyc

matthiasg, 2021-12-30:
Looked at the connectors again they seem fine as far as I can see

marzsman, 2022-01-22:
I'm having the same issue. I can't sync the RTC time.

```
pi@e-frame-pi:~ $ sudo systemctl status pisugar-server
● pisugar-server.service - pisugar-server
   Loaded: loaded (/lib/systemd/system/pisugar-server.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2022-01-21 23:47:25 GMT; 2min 15s ago
 Main PID: 329 (pisugar-server)
    Tasks: 6 (limit: 871)
   CGroup: /system.slice/pisugar-server.service
           └─329 /usr/bin/pisugar-server --config config.json --model PiSugar 2 (2-LEDs) --web /usr/share/pisugar-server/web --http 0.0.0.0:8421 --ws 0.0.0.0:8422 --tcp 0.0.0.0:8423 --uds /tmp/pisugar-server.sock

Jan 21 23:47:52 e-frame-pi pisugar-server[329]: [2022-01-21T23:47:52Z ERROR pisugar_server] Invalid slave address: 50
Jan 21 23:49:20 e-frame-pi pisugar-server[329]: Fri 21 Jan 23:49:20 GMT 2022
Jan 21 23:49:20 e-frame-pi pisugar-server[329]: hwclock: Cannot access the Hardware Clock via any known method.
Jan 21 23:49:20 e-frame-pi pisugar-server[329]: hwclock: Use the --verbose option to see the details of our search for an access method.
Jan 21 23:49:24 e-frame-pi pisugar-server[329]: [2022-01-21T23:49:24Z ERROR pisugar_server] Invalid slave address: 50
Jan 21 23:49:29 e-frame-pi pisugar-server[329]: [2022-01-21T23:49:29Z ERROR pisugar_server] Invalid slave address: 50
Jan 21 23:49:32 e-frame-pi pisugar-server[329]: [2022-01-21T23:49:32Z ERROR pisugar_server] Invalid slave address: 50
Jan 21 23:49:33 e-frame-pi pisugar-server[329]: [2022-01-21T23:49:33Z ERROR pisugar_server] Invalid slave address: 50
Jan 21 23:49:34 e-frame-pi pisugar-server[329]: [2022-01-21T23:49:34Z ERROR pisugar_server] Invalid slave address: 50
Jan 21 23:49:39 e-frame-pi pisugar-server[329]: [2022-01-21T23:49:39Z ERROR pisugar_server] Invalid slave address: 50
```
I'm using the Lite OS version
```
pi@e-frame-pi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
```

marzsman, 2022-01-22:
I did a reinstall following the instructions here
https://github.com/PiSugar/pisugar-power-manager-rs/issues/44

And now my RTC works

closed 2024-07-13 by github-actions[bot]