PiSugar/PiSugar #66: Provide STL or STEP for PiSugar boards and PiSugar battery

bgilreath, 2021-12-31:
The models can be low fidelity and dimensionally accurate.

This would allow others to integrate into cases of different designs than one provided.

I was not able to locate within the current documentation.

**Note:** This repo is mainly about the entire PiSugar project, cases and wiki. We usually process these issues once a month. For quick response about PiSugar products, please contact pisugar.zero@gmail.com. For software issues, please submit your issuse to https://github.com/PiSugar/pisugar-power-manager-rs/issues

JdaieLin, 2022-01-11:
Good suggestion. I will upload the pisugar STL models this weekend.

JdaieLin, 2022-01-18:
You can find the boards' STL here:

https://github.com/PiSugar/PiSugar/blob/master/model2/pisugar2Board.STL
https://github.com/PiSugar/PiSugar/blob/master/model3/pisugar3Board.STL
https://github.com/PiSugar/PiSugar/blob/master/model_pro/pisugar2pro.STL
https://github.com/PiSugar/PiSugar/blob/master/model_pro/pisugar3plus.STL

closed 2022-02-04 by bgilreath