PiSugar/PiSugar #70: pisugar 3

seespotruminate, 2022-01-13:
recently received version 3 plus, it has new mscl, sdat, sscl on underside. I am wondering for an embedded project what theses do? Also is there a way to use them or pi GPIO to wire a pushbutton switch for on/off?

JdaieLin, 2022-01-18:
The sdat, sscl are actually rpi's i2c bus. mscl, mdat are i2c driven by pisugar's chip, it's possible to extends the hardware feature in the future. 
If you need a external push button switch, you can use the pwr pad near the switch. We will offer a new feature to turn this switch into software mode.

seespotruminate, 2022-01-24:
Thanks! And the button pad is the solder point correct?

JdaieLin, 2022-02-04:
> Thanks! And the button pad is the solder point correct?

use the PWR solder pad and a GND

Tetsujinfr, 2022-03-19:
Can I use the standard power up rpi feature (connecting gpio3cand gdn) and use the ps3+ at thes same time?
It use to work with ps2pro, but with ps3+, so far, shutting down the rpi with some script turns power completely, and then it is impossible to wake up my system.
Btw, the only U can turn my ps3+ it to click the tiny round button on the back of the pcb (looks like the infamous activation button, but there is no documentation on this button nonetheless. Weird!

closed 2024-07-10 by github-actions[bot]