PiSugar/PiSugar #71: Installation Issue when configuring pisugar-server

ben8622, 2022-02-02:
When I am trying to install the software it eventually asks you to choose the PiSugar model in an interface similar to `sudo raspi-config`. When it reaches here I am unable to move up and down the selections using arrow keys which also stops me from hitting `<Ok>` and exiting the process. Hitting the arrow keys looks like the picture attached

Using a Raspberry Pi 4 with official Kali Linux for Raspberry Pi distribution. I can do `sudo raspi-config` and move the interface as normal. I have tried two different keyboards with the same results.
[image]

JdaieLin, 2022-02-04:
@fengyc

Logsleeve, 2022-04-19:
Did you ever get this figured out? I ran "sudo dpkg-reconfigure pisugar-server" and was able to select all of the prompts. However I never got the server to work on Kali. I also cannot update the firmware since its not for arm64. Did you ever get it running with Kali?

ben8622, 2022-04-19:
Nope lol, it wasn't that important for the project I was using so I just left it

mmider, 2023-11-17:
I have the same issue on my Pi Zero W running headless Debian Bookworm (am connected via ssh).

mmider, 2023-11-17:
> I have the same issue on my Pi Zero W running headless Debian Bookworm (am connected via ssh).

The issue I've been experiencing on a fresh install of Debian Bookworm appears to have resolved itself after updating the system (`sudo apt-get upgrade`).

closed 2024-07-10 by github-actions[bot]