PiSugar/PiSugar #72: PiSugar 3 issues

chickenherb, 2022-02-03:
**Note:** This repo is mainly about the entire PiSugar project, cases and wiki. We usually process these issues once a month. For quick response about PiSugar products, please contact pisugar.zero@gmail.com. For software issues, please submit your issuse to https://github.com/PiSugar/pisugar-power-manager-rs/issues

chickenherb, 2022-02-03:
Errors of PiSugar V3 with FW 1.05 and PiSugar Server V 1.6.4:

1. PiSugar V3 switches off unattended the power to the Rasperry Zero W without power-outage!
2. PiSugar V3 switches off unattended "battery_allow_charging: false" und never on again!
   This means that the batery will be drained and therefore the Rasperry Zero W wil shut down
   and power-off.
3. During restart the "input_protect: false" will be set, no matter what previous state it was.

JdaieLin, 2022-02-04:
@fengyc

closed 2024-07-10 by github-actions[bot]