PiSugar/PiSugar #75: Pisugar3 Plus docs and datasheet

Tetsujinfr, 2022-03-21:
Hi

I can not find much, if any, real documentation about the Pisugar3 Plus. How to use it? What are the board buttons and plugs functions?
E.g. what is the small round button on the back of the board? An "activate" bitton? If I do not press it, the board does not power up, really no convenient. Not reference of this button is to be found for the pisugar 3 plus.
What about the extra microUSB plug? Is it an alternative charging plug if one soes not want to use the USB-c plug? Is it another charging plug in parallel of the usb-c one? Is it a power source to power peripherals? No info.

Thanks for your help to one of your customer.

Tetsujinfr, 2022-03-24:
Anyone from pisugar?

JdaieLin, 2022-03-26:
The small round button on Pisugar3 is for reset. It seems abnormal if you have to press it before power up. Please contact our product support email for more detail.
The microUSB plug is an alternative charging plug.
More detail document please refer to:
https://github.com/PiSugar/PiSugar/wiki/PiSugar-3-Series

Tetsujinfr, 2022-03-26:
I have seen this page, it is not realky a documentatiin, rather a sales pitch of the features of the bord.
For instance, it boasts a "anti-mistaken touch button": but does not indicate where and what it is. Does it refer to the small round button in the back of the board? (Which was already present on pisugar pro2 and is a pain to deal with?)
It mentions a changeable i2c address: cool but how do you do it in practice? A code/instructions example would be more useful.
It mentions the rtc which can hold time about 1yr without battery: it may but when I had my battery ran out of juice, after restarting my rpi, but still being off internet connection, the rtc clock was significantly off from the real time.

I am considering reverting back to the pi sugar pro2 and using a separate i2c transmuxer to deal with i2c adresses conflicts I had with it. The glue of the magnet attached to the battery went loose, and when opening my prototype case I found the battery hanging with it power cable, the two magnets clamped on the board on there own... the temperature of the cpu somewhat spread to the magnet and got the glue to fail binding it. The magnet thing is a smart diy design but not robust enough for real projects I am afraid. I may have the same issue with the pisugar pro2 but at least now I know about it.

Tetsujinfr, 2022-03-26:
By the way, you may want to test the behaviour of the pisugar3 plus when using a power button connected on pin 5 and 6. For me, whan I shutdown my rpi, it does cut power fully on the board, hence impossible to wake it up by pressing the button (which connect pin 5 and pin 6).
Only way to get the power back it to press the infamous small round rest button that no one asked for and which is not documented on your wiki page yet.

Tetsujinfr, 2022-04-03:
Ok so, I have pligged my pisugar onto a different raspberry pi3b+ to test it more easily.
Below, I describe my setup and the scenario that I would like to happen, versus what is happening for real, so that you can tell me if it is expected or I am doimg something wrong.

Setup:
My pisugar3+ is connected to a power source via its micro usb power connector (the usb-c power connector is not connected to anything).
The raspberry micro usb pi power is NOT connected to any power source.
The pisugar3+ batterry is charged.
The rpi power leds are idle at that point, even if I press the pisugar3+ small power button. The only way I found to power up the rpi is to press the tiny very small circular button on the back of the pisugar3+ (after pushing the battery so that I can press the btn with my finger).
So I press this tiny round button and the rpi starts up.
Now, if I shutdown the rpi properly, via the OS main menu "shutdown" option, the completely shutsdown, the power leds turns off on the rpi pcb. So, I can not use a powerup button  to wake up the rpi, e.g. through the usual way of shorting pin 5 and 6 via a swith button.
If I press the small "pwr" of the pisugar3+ the rpi does not power up neither.
The only way for me to power up the rpi again, is to press the tiny round button on the back of the pisugar3+ : this is a nightmare to access and I do not see the point of having the "pwr" button on the pisugar3+ pcb then. The only thing the "pwr" button does, is to indicate the charging level of the battery, and to shut down the rpi if I press it for long (again, total power cut on the rpi board, all leds turned off on the rpi).

Expected behavior: to have the rpi led turned ON at all time, i.e. if I shut down the rpi through software, the rpi pcb is still under tension, its red power led is still ON and I can wake up the rpi via shorting pin 5 and 6.
If the pisugar3+ has been turned off via a long press of the "pwr" button, to be able to power it back ON bu pressing the "pwr" button again.

Please not that I did upgrade the pisugar3+ firmware with the latest version available as soon as I installed it. The upgrate was marked as succesfull.

fheslouin, 2022-04-04:
@Tetsujinfr did you look for anti mistaken touch function in 6. of https://github.com/PiSugar/PiSugar/wiki/PiSugar-3-Series ?
-> 1 short press followed by 1 long press (5s) to start up the board. 

I dislike this behavior but this is the way to start the board, you can disable this function by software but it is reset each time the battery is disconnected or empty :(

Tetsujinfr, 2022-04-04:
@fheslouin thanks for your comment.

Tbh what I need is to make sure the rpi pcb is always under tension so that I can turn it back on if it has been shutdown properly by software.
On the WebGUI of the pisugar, there is a small comment font size 2 where it mentions that if one wants to power the rpi through its power connector (which for me is the only way so far I can make sure the rpi pcb will stay under tension), and the pisugar through its power connector at the same time, then two different adapters have to be used, otherwise the pisugar will charge itself.
So I am setting up a second adapter.
I do not want my users to have to mess around with the tiny pisugar buttons (my project has the rpi in a closed box, so no access to the pisugar buttons anyways.

I thought the pisugar was a self sufficient UPS solution to cover all scenarios properly, but so far it is not for me.

mitpre, 2022-08-01:
@Tetsujinfr I'm also developing something, that would need the 'self sufficient UPS solution' without the need to access pisugar buttons. Did you maybe find a solution?

kjus, 2023-03-07:
So no datasheet? I'm also struggling with some features.

danielmorozoff, 2023-07-03:
@JdaieLin Do you guys have a component specs and datasheet ? Could you please share it.

Thanks

StephenGrable1, 2024-05-04:
@wlmh110 is the recommended documentation for the PiSugar 3 plus [here](https://github.com/PiSugar/PiSugar/wiki/PiSugar-3-Series#function-specifications)? or does that differ significantly from pisugar 3?

Tetsujinfr, 2024-05-04:
> @Tetsujinfr I'm also developing something, that would need the 'self sufficient UPS solution' without the need to access pisugar buttons. Did you maybe find a solution?

Sorry, did not see your question.
So I gave up on the pisugar products. Too messy to deal with, no proper doc or support, including here on github. It is for small diy projects and even so good luck. It is not robust for slightly challenging conditions, like in a case where temps would go in the 40C-45C, given the glue of the magnet just looses its properties. 

I did test the [waveshare HAT](https://www.waveshare.com/ups-hat-c.htm) for rpi zero, and works well, fyi.

I did not try their rpi 3-5 UPS product yet, but I would try that first if I was considering redoing a new prototype which requires power outage resiliency.

I am done with PiSugars I am afraid.

closed 2024-07-09 by github-actions[bot]