PiSugar/PiSugar #77: i2c not detected, PiSugar 3

bgilreath, 2022-03-24:
sudo i2cdetect -y 1 shows no devices on bus 1. PiSugar3 Plus is providing power to pi.

JdaieLin, 2022-03-26:
@wlmh110

jpkong, 2022-06-17:
Hi, guys.

I met the same issue also.
When I gave the cmd 'sudo i2cdetect -y 1' to RPi 4B with PiSuagr3+, it puts the below.
(I'm currently using Ubuntu 21.04 desktop for RPi)
`
rpi@rpi-desktop:~$ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
rpi@rpi-desktop:~$ 
`

Could anyone check on this?

wlmh110, 2022-06-17:
> sudo i2cdetect -y 1 shows no devices on bus 1. PiSugar3 Plus is providing power to pi.

Hi，what system is used?

wlmh110, 2022-06-17:
> Hi, guys.
>
> I met the same issue also. When I gave the cmd 'sudo i2cdetect -y 1' to RPi 4B with PiSuagr3+, it puts the below. (I'm currently using Ubuntu 21.04 desktop for RPi)
>
> ``` [...]

Under Ubuntu system, bus1 may not be available, and bus0 may be available. Please try:
i2cdetect -y 0

jpkong, 2022-07-12:
Hi, I checked it was an Ubuntu Desktop 21.04 version issue.
I don't know if the original OS image has the case or if some configs were wrong.
(I got the image from another engineer)

wlmh110, 2022-07-13:
> Hi, I checked it was an Ubuntu Desktop 21.04 version issue. I don't know if the original OS image has the case or if some configs were wrong. (I got the image from another engineer) So, I'm considering using Ubuntu 20.04 or 22.04. 22.04 is OK.

It should be that different systems lead to different I2C buses. Please try 0 bus or check the system information
i2cdetect -y 0

jpkong, 2022-07-13:
> It should be that different systems lead to different I2C buses. Please try 0 bus or check the system information i2cdetect -y 0

I tried to use 'i2cdetect -y 0' in the 21.04, but there was no 0 I2C bus.
Anyway, I'm going to 22.04 or 20.04.

Leviathan3DPrinting, 2023-01-01:
Also not showing up raspbian minimal

Maynardnaze, 2024-02-04:
Thank god for amazon.. mine did the same thing. no response from buttons, just power on instantly with no I2C value.

closed 2024-07-09 by github-actions[bot]