PiSugar/PiSugar #79: How to start PiSugar 2 Pro

tboonstl, 2022-03-29:
I know this sounds kind of basic, but I can't seem to start the battery pack.  

I've done the quickstart steps: Screwed the pack to the Pi 3, connected power to the pack, pressed the reset button for 2 seconds and then turn on.  I then installed the software and type the IP address:8421 and see the PiSugar dashboard.

But, there is no information there.  No battery level or anything except schedule wake-up and similar settings.

If I plug power to the PiSugar, the computer starts.  So, this seems to indicate that the connections between the battery and Pi are good.  Could the battery just be dead?

Any help would be greatly appreciated.

Tetsujinfr, 2022-04-03:
Hi, please provide a bit more details so that I can help. I feel your pain though, you pb seems kond of basic but no one from pisugar bothered to droo you a note in 5 days: not a good customer support!

Based on you description, it sounds your pisugar powers your pi, is that correct?
When you charge it, you should see a led turned on (blue led on my pro2), if you then unplug the charging cable, if your rpi still runs, it means your battery is working (at least a bit).
Now, if you do see the battery level in the browser on url 127.0.0.1:8421 then it may be due to a conflict with another device on your i2c channel. To check this out, in a terminal type the below instruction and share its outout here in this thread:
I2cdetect -y 1

You should see address 75 popup in the results, for the pisugar pro2.

wlmh110, 2022-04-06:
Hi， This sounds like a product failure. It would be best if you could shoot an operation video for us.
Customer service email: pisugar.com zero@gmail.com

JdaieLin, 2022-05-22:
> Hi， This sounds like a product failure. It would be best if you could shoot an operation video for us. Customer service email: pisugar.com [zero@gmail.com](mailto:zero@gmail.com)

pisugar.zero@gmail.com

closed 2024-07-08 by github-actions[bot]