PiSugar/PiSugar #80: Pisugar 2 pro does not maintain rpi pcb under tension when rpi is powered off via software

Tetsujinfr, 2022-04-03:
I jave a pisugar pro2 and a rpi3b+.
I keep my pisugar power switch always on ON, and I do not want to use any of the pisugar tiny buttons for power on/off operations, I use my own buttons for that, which are attached to the rpi3b+ gpios.

Ok so after testing for a while, I realise that when I shut down my raspberry properly via the OS shutdown menu option, my pisugar 2 pro does not maintain the raspberry pcb under tension and everything is off (rpi pwr red led is off). So at that point, impossible to wake the rpi on with a wakeup button shorting pin 5 and 6 as usual. Not good for me!

Now, if I uninstall the power management  software that you provide with the pisugar suite of boards, then if I shutdown the rpi properly, this time the pisugar maintains the rpi pcb under power!!

So, is that a problem with your poweroff server code?
Am I the only one who want to be able to power up my rpi after having shut it down properly ?
Thanks!

JdaieLin, 2022-04-06:
@wlmh110

wlmh110, 2022-04-06:
Hi,
You can try to remove the PiSugar-Poweroff and open the fuction called "on power restore"

Tetsujinfr, 2022-04-06:
@wlmh110 thanks. Yes I have done this and that definitely helps.
I had to add a second adapter, so that I an sure that the rpi is always connected to power when it is available via main, so that it can be woke up by a button (not one of the pisugar buttons).
So as long has my pisugar battery does not get completely depleted, this should work.
I really recommend that in a future version, you have an option the user can choose whereby if the rpi is powered off, the pisugar keeps feeding power to the rpi pcb, that would have saved me a second charger. Closing this issue, kind of.

closed 2022-04-06 by Tetsujinfr