PiSugar/PiSugar #82: Pi Sugar S plus case for RPi 3B/3B+

wcraigtrader, 2022-04-16:
I just received my Pi Sugar S Plus, and I wanted to 3D print a case for it.  Alas, Pi 4s are unobtanium, so I'm using a Pi 3B+ for my project.

There's no specific set of models for the Pi Sugar S Plus. The closest seems to be the `model_pro` models, and they only support the Pi 4B, not the Pi 3B/SB+.

JdaieLin, 2022-04-19:
Yes the current pro model is only for 4B, we will provide the 3B parts in this month.

wcraigtrader, 2022-04-19:
Thanks. It would be helpful if the models will be printable by a conventional FDM printer.

JdaieLin, 2022-05-03:
Hi, the S Plus parts for 3B had been uploaded. 2 Pro cases are the same as S Plus.
The parts you need:

[pro - top.STL](https://github.com/PiSugar/PiSugar/blob/master/model_pro/pro%20-%20top.STL)
[pro - pro_shell.STL](https://github.com/PiSugar/PiSugar/blob/master/model_pro/pro%20-%20pro_shell.STL)
[pro - bottom.STL](https://github.com/PiSugar/PiSugar/blob/master/model_pro/pro%20-%20bottom.STL)
[pro - port-panel-3b-2p-1.STL](https://github.com/PiSugar/PiSugar/blob/master/model_pro/pro%20-%20port-panel-3b-2p-1.STL)
[pro - usb-panel-3b-1.STL](https://github.com/PiSugar/PiSugar/blob/master/model_pro/pro%20-%20usb-panel-3b-1.STL)
[button_switch_batch.STL](https://github.com/PiSugar/PiSugar/blob/master/model_pro/button_switch_batch.STL)

JdaieLin, 2022-05-03:
standalone button & switch
[pro - button.STL](https://github.com/PiSugar/PiSugar/blob/master/model_pro/pro%20-%20button.STL)
[pro - switch.STL](https://github.com/PiSugar/PiSugar/blob/master/model_pro/pro%20-%20switch.STL)

closed 2022-05-03 by wcraigtrader