PiSugar/PiSugar #96: Question: Pisugar 3 power on

gimzmoe, 2022-08-31:
I have a pisugar v3 that appears to be functioning. 

I programmed the 2-click shutdown on the button adjacent to the usb-c power input.
To power on, I assume you press and hold the same button however that never appears to work, I've tried 2-10+ seconds
The unit does power on when I plug in a usb-c to charge it.

Is there a way to power on from an off-state without external power?

gimzmoe, 2022-09-01:
I kept digging and I found a clue when I eventually found the PCB diagram from https://github.com/PiSugar/PiSugar/wiki/PiSugar-3-Series which informed what the buttons were (I inferred it backwards) 
To power on you hold and watch the LEDs count up to boot.
If I'm not the only oblivious one, hopefully this helps someone else.

emilianos, 2023-05-19:
@gimzmoe I was having the same issue and found this: https://github.com/PiSugar/PiSugar/issues/115
**To power on with the side button, it requires a short press followed by a long press**

This should be way more visible in the documentation imo.

closed 2024-06-24 by github-actions[bot]