PiSugar/PiSugar #101: Pisugar 3 Plus External on/off switch solder pad confusion.

kentheckel, 2022-09-27:
Hello!

I sent an email to the provided but haven't heard anything.

I am trying to add an external power on/off to the pisugar 3 plus. 

I am looking at your [documentation](https://github.com/PiSugar/PiSugar/wiki/PiSugar-3-Series) and noticed an extra solder pad on my pisugar 3 plus. It's labeled "BAT" and it sits between "BTN"(9) and "PWR"(10). 

In my testing when the pisugar is OFF the "BTN" pad will light up my external power button and the blue LED on the pisugar 3 will also light up indicating it sees the button and reacts to it. But it will not turn on the pisugar.

When the Pisugar 3 plus is ON and I attach the button to the "PWR" pad it lets me turn off the battery just fine. I am using the anti-mistake shutdown. 

How would you recommend I set up this power button? 

Thanks for the help

Pic of my pisugar3 plus attached
[image]

JdaieLin, 2022-10-29:
@wlmh110

chipspencer, 2023-06-13:
Any movement on this issue? I too, am interested in properly adding an external power button. This would be a great addition to the PiSugar 3 documentation.

lambtor, 2023-12-05:
is the pad for the external power control mean to be used with a button, or with a slide switch?

closed 2024-06-23 by github-actions[bot]