PiSugar/PiSugar #102: [Question] Does Pisugar 3 Plus support Raspberry Pi Zero 2W?

eqikkwkp25-cyber, 2022-09-28:
Does Pisugar 3 Plus support Raspberry Pi Zero 2W? And if not what is the reason?

Ra2-IFV, 2022-10-02:
It depends on which OS you use, I think.

eqikkwkp25-cyber, 2022-10-03:
> It depends on which OS you use, I think.

@Ra2-IFV Do you have any reference or sources for this statement? And with which OS it works? My preferred OS is Raspian as for many others.

Ra2-IFV, 2022-10-13:
> @Ra2-IFV Do you have any reference or sources for this statement? And with which OS it works? My preferred OS is Raspian as for many others.
> 
> My guess that the behavior is the same with a Pisugar 2 Pro 5000mah where the Pi Zeros are not listed as supported boards too.

Well, I think the only difference between Pi Sugar 3 Plus and Pi Sugar 3 is... They just made it bigger.
Since the GPIO pinouts of Rpi4 and Rpi Zero are TOTALLY the same, there is no reason why Pi Sugar 3 should not be compatible with Rpi Zero.

Ra2-IFV, 2022-10-13:
Of course if your OS supports I2C, then it should support Pi Sugar

eqikkwkp25-cyber, 2022-10-29:
In the meantime i own a PiSugar 2 Plus and it does not seem to work with a Pi Zero 2. See issues and comments within https://github.com/PiSugar/pisugar-power-manager-rs

closed 2024-06-22 by github-actions[bot]