PiSugar/PiSugar #104: Init battery issue causing pi zero 2 W to not boot

paulbuzzby, 2022-10-25:
My Pi Sugar has been working really well until 3 days ago when it no longer fully boots the pi. After managing to get into the logs (i'm not a great linux user) I can see the following messages.

Oct 25 10:41:10 raspberrycal pisugar-server[356]: [2022-10-25T09:41:10Z INFO  pisugar_core] Init battery...
Oct 25 10:41:10 raspberrycal pisugar-server[356]: [2022-10-25T09:41:10Z ERROR pisugar_server] Require initialization
Oct 25 10:41:10 raspberrycal pisugar-server[356]: [2022-10-25T09:41:10Z ERROR pisugar_server] Required initialization
Oct 25 10:41:11 raspberrycal pisugar-server[356]: [2022-10-25T09:41:11Z ERROR pisugar_server] I/O error: Remote I/O error (os error 121)

These are repeated through the log but these are the 4 distinct messages that I get.

The project I am using this on is mounted to a fridge door so i wondered if the pugo pins had moved. I have re seated the piSugar but get the same behaviour. In that the Pi will start to boot and then just power off after a few seconds.

I can access the web interface for the PiSugar if i power the pi directly. I can see the RTC is running and can sync it but no battery information is displayed. 
It is like the PiSugar cannot detect the battery.

Any help would be greatly received. Suspect some sort of hardware issues as the pi has had no software updates since I finished the project a few months ago.

malcolmosh, 2023-10-25:
Hi Paul - I noticed you're using the same setup as I was - a 7.5 e-ink HAT plus the PiSugar2. I'm also getting the same problem ; the dashboard loads but without battery information. Do you remember how you solved it ?

closed 2023-10-25 by paulbuzzby

malcolmosh, 2023-10-25:
Thanks! I'll look into it. I had limited success a few minutes ago after reinstalling the Pi's OS and rebooting - so it may yet work.